WEC-Sim/WEC-Sim #1302: An Error using wecSim

HareendraAmaratunge, 2024-09-02:
I tried wecSim to get some mooring results by modeling a new hemisphere design. I used WAMIT sphere for bemio. And I drove SLX file. and made up a wecsim input file by slightly differentiating the RM3MoorDyn input file. But in the end I came up with an error called "Index must not exceed 1. I ll atach the error. please help. 

[image]

Regards.

kmruehl, 2024-09-04:
@HareendraAmaratunge can you please amend this issue to use the issue template? Thank you.

HareendraAmaratunge, 2024-09-04:
I Will.

HareendraAmaratunge, 2024-09-04:
Ok. I will

regards.

HareendraAmaratunge, 2024-09-07:
rgarding I amended this issue to use the issue template as you said.

closed 2024-09-07 by HareendraAmaratunge