WEC-Sim/WEC-Sim #1303: Wec Sim Error

HareendraAmaratunge, 2024-09-04:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
_Identify which project this issue is associated with._

**Describe the issue**
_A clear and concise description of the issue being encountered and any necessary context._

kmruehl, 2024-09-04:
@HareendraAmaratunge can you please upload a ZIP of your WEC-Sim model so that we can help debug the issue? Thank you.

HareendraAmaratunge, 2024-09-05:
Please help me. I m working on a research that has to define a mooring
system for a WEC.
thanks for helping.

HareendraAmaratunge, 2024-09-05:
Please help me. I m working on a research that has to define a mooring
system for a WEC.
Thanks for helping.

[Hemisphere.zip](https://github.com/user-attachments/files/16884076/Hemisphere.zip)

Regards.

HareendraAmaratunge, 2024-09-07:
Any Updates on this? 

I tried to model the requirement using NEMOH sphere code in the BEMIO. it too gives an error like this.

[image]

Please help. 
Regards.

jtgrasb, 2024-10-14:
Hi @HareendraAmaratunge please let us know if this resolves your issues.

HareendraAmaratunge, 2024-10-20:
Hi.
Thank You for responding to me. Now I came up with a simple SLX design
without a constraint. Now that gives me results. And I changed the research
topic. Now there is no need for mooring. Now I m trying to make an
optimized wave energy converter for the Sri Lankan sea condition. I need to
know can I have the Energy of a wave through this software? Please help.

Regards!

jtgrasb, 2024-10-21:
@HareendraAmaratunge If you are trying to access the potential power of the waves, you can use the variable `waves.power`. To access the mechanical power absorbed by the PTO, you can use `output.ptos.powerInternalMechanics`. For examples of creating power matrices and other information, please refer to our [documentation](https://wec-sim.github.io/WEC-Sim/main/user/advanced_features.html#multiple-condition-runs-mcr).

jtgrasb, 2024-10-29:
It seems like your issues have been resolved, so I am closing this issue. Please comment or open a new issue if you have more questions.

HareendraAmaratunge, 2024-10-31:
I am having an error called Plotspectrum requires phase array system
toolbox. how can I obtain this?

jtgrasb, 2024-10-31:
Please first ensure WEC-Sim source and all its subfolders are properly added to the path, and you are using `waves.plotSpectrum();`, which is a function from the WEC-Sim wave class. If still giving you an error, please provide more information on the code you are trying to run, the error message, WEC-Sim version, MATLAB version, etc.

HareendraAmaratunge, 2024-10-31:
Here I submitted my white window down below. I need to plot the spectrum
and ran the function which is provided in the waves.functions. Then I come
up with the phase array problem all the time. Also I think i added all the
subsidiaries of the source to the path. I attached a photo of it too.

jtgrasb, 2024-10-31:
The images did not come through. You may need to respond on the Github website to attach images.

HareendraAmaratunge, 2024-10-31:
[image]
[image]

Sorry for the inconvenience. I uploaded the photos again

kmruehl, 2024-10-31:
@HareendraAmaratunge `plotSpectrum` is a method within the `waveClass`, so to use it you'll need to use the command `wave.plotSpectrum`. Please refer to the RM3 for an example of how to use this method:
https://github.com/WEC-Sim/WEC-Sim/blob/1cdd46313bf52a7cc086db1f57233dcf26ae514b/examples/RM3/userDefinedFunctions.m#L6 for an example of how to use this method.

HareendraAmaratunge, 2024-11-02:
the code you supplied just before was worked. Heave responses were plotted. But what I need here is to plot the generated power graph. 
I tried this command several times for RM3 model but always it shows "'plotSpectrum' requires Phased Array System Toolbox."

 % plotSpectrum();
output = plotSpectrum();                             
            m0 = trapz(output.omega,output.spectrum);
            HsTest = 4*sqrt(m0);
            [~,I] = max(abs(output.spectrum));
            wp = output.omega(I);
            TpTest = 2*pi/wp;
            
            figure
            plot(output.omega,output.spectrum,'s-')
            hold on
            line([wp,wp],[0,max(output.spectrum)],'Color','k')
            xlim([0 max(output.omega)])
            title([output.spectrumType, ' Spectrum, T_p= ' num2str(TpTest) ' [s], '  'H_m_0= ' num2str(HsTest), ' [m]'])
            if strcmp(output.spectrumType,'JS') == 1
                title([output.spectrumType, ' Spectrum, T_p= ' num2str(TpTest) ' [s], H_m_0= ' num2str(HsTest), ' [m], gamma = ' num2str(output.gamma)])
            end
            xlabel('Frequency (rad/s)')
            ylabel('Spectrum (m^2-s/rad)');

please help.

jtgrasb, 2024-11-05:
Hi @HareendraAmaratunge. It is difficult to pin down where the error is arising. Can you please describe your workflow step by step so we can identify what exactly is causing the issue?

jtgrasb, 2024-11-20:
@HareendraAmaratunge Are there any updates on your issue?

HareendraAmaratunge, 2024-11-21:
hii.
Sorry for being late for the reply. I ve been working on my thesis these
days. I found  a code to generate the power plot and have the energy
generated.

Regards!

closed 2024-11-21 by jtgrasb

HareendraAmaratunge, 2024-11-23:
Awesome, thanks for the update!
And I'll be willing to do another research on this. And I'll let you know.
Thanks for keeping helping me for all along by now.

Regards!