WEC-Sim/WEC-Sim #1310: [Developer Issue] Add Variable Hydrodynamics Documentaiton

kmruehl, 2024-09-06:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Project**
Variable Hydrodynamics https://github.com/WEC-Sim/WEC-Sim/pull/1248

**Describe the issue**
Variable Hydro PR has been merged, but we need to add documentation
- [ ] add documentation

akeeste, 2024-10-30:
Removing this card and combining it with an overarching project on variable hydro: https://github.com/orgs/WEC-Sim/projects/12/views/1?filterQuery=variable&pane=issue&itemId=78810185

akeeste, 2025-02-05:
This issue is already resolved by #1343. if there are further questions, we can add more information

closed 2025-02-05 by akeeste