WEC-Sim/WEC-Sim #1311: Update Applications

kmruehl, 2024-09-06:
**Applications**
* Target future release
* Discussed on 7/31

**ACTIONS:** Develop new applications and tests for the following features:
- [ ] update WECCCOMP test, which one(s) should we test? Currently WECCCOMP Fault and WECCCOMP MPC are tested
- [ ] wecSimPCT 
- [ ] other new applications?

**7/31 Update**
Stretch goal for v6.1

kmruehl, 2025-04-10:
This is addressed in PR https://github.com/WEC-Sim/WEC-Sim_Applications/pull/82

kmruehl, 2025-05-30:
@kmruehl add this citations for Load Mitigating Controls:

Forbush, D. D., Bacelli, G., & Coe, R. G. (2022). Self-tuning, load-mitigating feedback control of a 3-DOF point absorber. International Marine Energy Journal, 5(1), 23–35. https://doi.org/10.36688/imej.5.23-35

kmruehl, 2025-06-11:
@salhus can you please provide the citations for the two cases?

kmruehl, 2025-06-11:
From @dforbush2 for OWC:

https://www.researchgate.net/publication/363010817_A_post-processing_technique_for_removing_%27irregular_frequencies%27_and_other_issues_in_the_results_from_BEM_solvers

dforbush2, 2025-06-11:
For OWC applications case:
WAMIT manual for methods used

For the BEM cleanup methods
https://www.researchgate.net/publication/363010817_A_post-processing_technique_for_removing_%27irregular_frequencies%27_and_other_issues_in_the_results_from_BEM_solvers

closed 2025-07-11 by kmruehl