WEC-Sim/WEC-Sim #1314: [Developer Issue] Comments for electricGeneratorEC are Incorrect

nmgray, 2024-09-07:
**Project**
Wec-Sim

**Describe the issue**
The comments for electricGeneratorEC are incorrect in WEC-Sim/source/objects/ptoSimClass.m. They are currently:
`        electricGeneratorEC (1,1) struct        = struct(...                % Electric Machine - Equivalent Circuit
            'Ra',                               0.1,...                     % [ohm]       Initial displacement of the piston measured from port A
            'La',                               0.1,...                     % [H]         Piston Area, side A
            'Ke',                               0.1,...                     % [V/(rad/s)] Back emf constant
            'Jem',                              0.1,...                     % [kg*m^2]    Bulk modulus of the hydraulic fluid
            'bShaft',                           0.1,...                     % [N*m*s]     Bulk modulus of the hydraulic fluid
            'currentIni',                       0.0,...                     % [A]         Electric motor current initial value
            'wShaftIni',                        0.0)                        % [rpm]       Shaft speed initial value`

The notes for Ra, La, Jem, and bShaft seem to be copy pasted from a piston element. They should be:
`        electricGeneratorEC (1,1) struct        = struct(...                % Electric Machine - Equivalent Circuit
            'Ra',                               0.1,...                     % [ohm]       Armature resistance
            'La',                               0.1,...                     % [H]         Armature inductance
            'Ke',                               0.1,...                     % [V/(rad/s)] Back emf constant
            'Jem',                              0.1,...                     % [kg*m^2]    Rotor moment of inertia
            'bShaft',                           0.1,...                     % [N*m*s]     Rotor viscous drag
            'currentIni',                       0.0,...                     % [A]         Electric motor current initial value
            'wShaftIni',                        0.0)                        % [rpm]       Shaft speed initial value`

It may be worthwhile to verify other PTOSim elements for correctness.

jleonqu, 2024-09-20:
@nmgray Thanks for pointing this out. I have revised the comments for the PTO-Sim class and I have updated the ones that are incorrect. I have submitted PR #1330  to solve this issue.

jleonqu, 2024-10-10:
PR #1330 has been merged, so I am closing this issue.

closed 2024-10-10 by jleonqu