WEC-Sim/WEC-Sim #1315: modeling a two body Pitch-type WEC

Mohamad-Namaki, 2024-09-10:
**Cite the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) case in question**  
I am working on modeling a pitch-type WEC using WEC-Sim and Ansys/AQWA as the BEM solver. I couldn't generate the required `.stl` file directly from Ansys/AQWA, so I used SolidWorks to create the `.stl` files.

**Describe the bug or issue**  
The mesh in the `.stl` file generated by SolidWorks is different from the mesh used in Ansys/AQWA for the Boundary Element Method (BEM). I am concerned that this discrepancy in mesh geometry might cause issues when interfacing with WEC-Sim or impact the accuracy of my simulations. It is unclear whether this will create problems during the simulation or if the difference in mesh formats can be resolved within WEC-Sim or Ansys/AQWA. 

**To Reproduce**  
Steps to reproduce the behavior:
1. Attempt to export the `.stl` file from Ansys/AQWA.
2. Failing that, create a similar geometry in SolidWorks and export it as `.stl`.
3. Compare the mesh structure between the SolidWorks-generated `.stl` file and the mesh used by Ansys/AQWA for BEM calculations.
4. Notice the discrepancy in mesh structure and format.

**Expected behavior**  
I expected that using an `.stl` file from SolidWorks would be compatible with WEC-Sim and Ansys/AQWA without issues. I assumed that the mesh used in SolidWorks would not significantly impact the simulation if it maintained the same geometry as the one used in Ansys/AQWA.

**Screenshots**  
N/A

**Workstation Information (this section MUST BE completed):**
- OS: Windows 11
- MATLAB/Simulink Release: R2022b
- WEC-Sim Release: v4.1

**Additional context**  
Would the difference in mesh structures between SolidWorks and Ansys/AQWA cause any significant issues in WEC-Sim simulations? How can I ensure the compatibility between the `.stl` files and the BEM solver used in Ansys/AQWA?

MShabara, 2024-09-10:
Hi @Mohamad-Namaki 

The mismatch between the .stl and the mesh file used to generate the hydrodynamic forces will not have an effect on WEC-Sim solution. The stl file in WEC-Sim plays two roles:
1. Visualization.
2. The calculation of the non-linear FK forces is the `body(i).nonlinearHydro` feature is enabled.

I hope this is helpful and let us know if you have any more questions.

Mohamad-Namaki, 2024-09-11:
Hi @MShabara,  
Thank you for your response!

I have a couple of follow-up questions:

1. You mentioned that if the `body(i).nonlinearHydro` feature is enabled, the `.stl` file should represent the actual geometry of the body. Does that mean the `.stl` file should also have the same meshing as the mesh used in the BEM solver (e.g., Ansys/AQWA)? If so, how should I export or generate an `.stl` file from Ansys/AQWA with a mesh that is compatible with WEC-Sim? Currently, I’m using SolidWorks to create the `.stl` file, but the mesh structure is different from the one used in Ansys/AQWA.

2. Another issue I’m facing is that I want to modify the values of `body(i).centerGravity`, `body(i).mass`, and `body(i).inertia` for a hydrodynamic body whose BEM calculations were performed in Ansys/AQWA. Specifically, I want to assign values different from those used in the Ansys/AQWA simulation. However, I receive a warning when I do this, and I’m concerned that it might affect the accuracy of my analysis. Can adjusting these parameters independently of the BEM results lead to inaccuracies, or is there a safe way to change them without corrupting the analysis?

I would appreciate any advice or clarification on these points!

MShabara, 2024-09-12:
@Mohamad-Namaki  

1.    No, the STL files do not have to match. However, when using STL files for the non-linear hydro calculations, please keep in mind the following:
         The function assumes that the STL file is imported with its center of gravity (CG) at (0,0,0).
         The triangular mesh in the STL file must be fine enough to accurately calculate the forces. This can be managed in SolidWorks through the STL export parameters.

2.    The current version of WEC-Sim uses only the values provided in the .h5 file. However, we are considering updating it to accept values from the wecSimInputFile.m. 
      The WEC-Sim team encourages community contributions to enhance the source code, and we would fully support you if you were to make this modification and submit a pull request. We would review it and, if everything is in order, accept the changes. We can also offer you guidance throughout the PR process if needed.

In the meantime, you can impose new values by updating the hydro structure in BIMIO before writing the .h5 files. The variables body(i).centerGravity, body(i).mass, and body(i).inertia are design parameters that you can modify based on your engineering judgment. But keep in mind:

1. Changing the center of gravity or the moment of inertia will affect the metacentric height, which could impact the stability of your system.
2. Modifying the system mass will alter the free surface location, potentially requiring new BEM runs.

I encourage you to use BEMRosetta to evaluate the effects of some of the changes you want before running WEC-Sim. You can download the executable from [this link](https://github.com/BEMRosetta/BEMRosetta/releases).

Mohamed Shabara

Mohamad-Namaki, 2024-09-14:
Hi @MShabara,  
Thanks for your detailed explanation!

The wave energy converter system I’m simulating in WEC-Sim involves two bodies: a hydrodynamic shell body (PitchFloat) and a non-hydrodynamic internal body (Pendulum). The system includes two rotational PTOs—one between the shell and a fixed point, and another inside the shell between the Pendulum and the shell. The combined mass and center of gravity (CoG) of the entire system (PitchFloat + Pendulum) are fixed, but the individual masses of the two bodies vary across different simulation runs.

I’m using AQWA as the BEM solver for frequency-domain analysis, and my goal is to perform several runs with different mass distributions to find the optimal ratio of total system mass for each body. I’ve carefully considered the points you raised regarding system stability, mass, and CoG, since the total mass and CoG are constant across all simulations.

My main question is:  
Can I modify the `body(i).centerGravity`, `body(i).mass`, and `body(i).inertia` values for each body in the `wecSimInputFile.m` so that they are different from those defined for the hydrodynamic bodies in AQWA and saved in the `.h5` file generated by BEMIO? Will this cause any issues or conflicts during the simulation, given that the total system mass and CoG remain constant but the individual properties of each body change?

Also, I’m encountering another issue. I’ve shared my `wecSimInputFile.m` code, an image of the WEC, and the Simulink model. I keep getting an error (I’ve attached a screenshot of the error), and I’m unable to resolve it. Any guidance you can provide on this would be much appreciated.

simulink model and the WEC:

[image]

`wecSimInputFile.m` code:

[image]

a picture of the message and the error printed in MATLAB command window:

[image]

MShabara, 2024-09-16:
Hi @Mohamad-Namaki 

Here are the answers to your questions:

1. > Can I modify the body(i).centerGravity, body(i).mass, and body(i).inertia values for each body in the wecSimInputFile.m so that they are different from those defined for the hydrodynamic bodies in AQWA and saved in the .h5 file generated by BEMIO? Will this cause any issues or conflicts during the simulation, given that the total system mass and CoG remain constant but the individual properties of each body change?

Changing these values will not cause issues with WEC-Sim or conflicts during the simulation.
The current version of WEC-Sim allows the user to set user defined values for `body(i).centerGravity`, and `body(i).mass`. I made [PR#1324](https://github.com/WEC-Sim/WEC-Sim/pull/1324) to allow the users to specify user defined values for `body(i).centerGravity` and `body(i).centerBouyancy` within the `wecSimInputFile`.

2. > Also, I’m encountering another issue. I’ve shared my wecSimInputFile.m code, an image of the WEC, and the Simulink model. I keep getting an error (I’ve attached a screenshot of the error), and I’m unable to resolve it. Any guidance you can provide on this would be much appreciated. 

It seems that the `clear` command is used in `wecSimInputFile.m`, which clears the MATLAB workspace. When `wecSim` is typed in the command window to run WEC-Sim, there are some background operations executed before `wecSimInputFile.m` is run.

Removing the `clear` command will prevent unintended clearing of important variables required during the execution of WEC-Sim, ensuring that the necessary variables remain intact when the wecSimInputFile.m script is executed.

Mohamed Shabara

Mohamad-Namaki, 2024-09-17:
Dear @MShabara,  
Thank you so much for your time and thorough explanations! The previous error has now been resolved, and I appreciate your help with that.

However, I’m facing another issue. As shown in the message and screenshot I previously shared, the MATLAB command window shows the "List of Body: Number of Bodies" as 1, even though I’ve defined 2 bodies—a hydrodynamic (PitchFloat) and a non-hydrodynamic (Pendulum) body. I suspect there may be an issue with how I’ve defined the bodies, but I’m not sure where the problem lies.

For example, regardless of what I set for `body(1).name`, it won’t accept my custom string and defaults to naming it `body1`. After running the `wecSimInputFile.m` file, I can see in the visualization that the two bodies are connected rigidly without the expected relative angular motion. I’ve already defined an angular PTO between these two bodies, so I’m unsure why the interaction is not working as intended.

Could you please help me identify where the problem might be in my body definitions or PTO setup?

MShabara, 2024-09-17:
@Mohamad-Namaki

You’ve raised some excellent points! Here are the answers to your questions:

 1.  **List of Bodies**
> ... As shown in the message and screenshot I previously shared, the MATLAB command window shows the "List of Body: Number of Bodies" as 1, even though I’ve defined 2 bodies—a hydrodynamic [...]

There’s nothing wrong with your body definitions. The current version of WEC-Sim only displays the number and details of hydrodynamic bodies. I’ve added a few commits to [PR#1324](https://github.com/WEC-Sim/WEC-Sim/pull/1324) that include printing the number of non-hydrodynamic bodies and their properties as well.

2. **Custom Body Name**
> For example, regardless of what I set for body(1).name, it won’t accept my custom string and defaults to naming it body1. ...

The current WEC-Sim version doesn’t consider the user-defined body name, instead using the name specified in the .h5 file. I’ve added a commit to [PR#1324](https://github.com/WEC-Sim/WEC-Sim/pull/1324) that prioritizes the user-defined name over the .h5 file name when specified.

3. **Angular PTO Not Functioning as Expected**
> After running the wecSimInputFile.m file, I can see in the visualization that the two bodies are connected rigidly without the expected relative angular motion. I’ve already defined an angular PTO between these two bodies, so I’m unsure why the interaction is not working as intended.

I’m not sure what’s causing this issue without seeing the model. Have you tried setting the PTO damping coefficient to zero and checking how the motion looks? Your current damping coefficient is set to 2.3e4, which may be too high for the pendulum size.

4. Also, I noticed that your moment of inertia values for the two bodies have Ixx = Iyy = Izz, which might not be accurate. You may want to double-check these values using Solidworks or a similar tool.

Mohamad-Namaki, 2024-09-20:
Dear @MShabara,

Thank you for your detailed response. I’m currently encountering some challenges regarding the units of angular inputs in relation to angular PTOs and constraints within `wecSimInputFile.m`. While I understand that the angular outputs are in radians, I’m uncertain whether the inputs should be provided in radians or degrees. Consequently, I’m unsure of the correct unit for input damping in angular PTOs — should it be expressed in N/(rad/s) or N/(deg/s)? I believe this may be contributing to the unexpected results I’m seeing.

Additionally, I would like to set an initial relative angular position using the `pto` class. Is the correct feature to use `pto(i).equilibriumPosition`? Furthermore, when setting `pto(i).hardStops.upperLimitSpecify`, should the value be provided in radians or degrees, and could you kindly explain how the hard stop mechanism functions?

Another issue I’m facing relates to the properties of non-hydrodynamic bodies. When defining the center of gravity (CoG), the location of the non-hydrodynamic body changes unexpectedly. Could you clarify how to correctly utilize the `body(i).initial.displacement` property for non-hydro bodies? Additionally, when inputting the moment of inertia, should it be calculated relative to the input CoG or from another point?

I would greatly appreciate your insights on these matters.

MShabara, 2024-09-20:
@Mohamad-Namaki 

1. > ... I’m uncertain whether the inputs should be provided in radians or degrees. Consequently, I’m unsure of the correct unit for input damping in angular PTOs — should it be expressed in N/(rad/s) or N/(deg/s)? I believe this may be contributing to the unexpected results I’m seeing.

The inputs are radians, please check the comments in line 46 of the [body class](https://github.com/WEC-Sim/WEC-Sim/blob/main/source/objects/bodyClass.m). Also check [Nathan's response](https://github.com/WEC-Sim/WEC-Sim/issues/480#issuecomment-789115860) for more insights on this issue.

The rotational PTO stiffness and damping coefficients are in [Nm/rad]  [Nsm/rad], you can use the [OSWEC model](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/e8cacc3364fbfbc84c5c6e60a3510b50e74aec90/PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO) for more details about setting up an initial rotation for the body.

2. > Additionally, I would like to set an initial relative angular position using the pto class. Is the correct feature to use pto(i).equilibriumPosition? Furthermore, when setting pto(i).hardStops.upperLimitSpecify, should the value be provided in radians or degrees, and could you kindly explain how the hard stop mechanism functions?

Yes you can use `pto(i).equilibriumPosition` for that purpose. Noting that the units for this should be in degrees not radians as mentioned in the comments of line 33 in the [pto Class](https://github.com/WEC-Sim/WEC-Sim/blob/main/source/objects/ptoClass.m).
The `upperLimitSpecify` is an 'on' and 'off' variable, please check these Issues #1140 and #574 to get insights on setting up `pto(i).hardStops` in WEC-Sim.

3. > Another issue I’m facing relates to the properties of non-hydrodynamic bodies. When defining the center of gravity (CoG), the location of the non-hydrodynamic body changes unexpectedly. Could you clarify how to correctly utilize the body(i).initial.displacement property for non-hydro bodies? Additionally, when inputting the moment of inertia, should it be calculated relative to the input CoG or from another point?
 
I think [Nathan's response](https://github.com/WEC-Sim/WEC-Sim/issues/480#issuecomment-789115860) mentioned in my answer to question 1 should have answered the first part of this question. 
Yes, the moment of inertia should be relative to the C.G location.

Mohamad-Namaki, 2024-09-24:
Dear @MShabara,

Thank you again for your clarification. I’ve been experimenting with the `pto(i).equilibriumPosition` parameter, referring to the OSWEC example. However, regardless of the value I assign, I haven't observed any impact on the results, and the initial relative angular position doesn't seem to be applied. It appears that setting the initial angular position would be more intuitive if it could be defined within `pto(i).initial`, as assigning initial displacement for an angular PTO seems less meaningful.

Additionally, I’ve encountered a recurring issue with hydrodynamic bodies. No matter what value I assign to `body(i).centerGravity`, I receive the following warning, and when I inspect the body class properties, the centerGravity value remains unchanged:

```
Initializing the Simulation Class...
Case Dir: D:\NREL\WECSim\GitHub\WEC-Sim\examples\PitchFloat 
Warning: Center of gravity and center of buoyancy are
overwritten by h5 data for hydro bodies. 
In bodyClass/checkInputs (line 231)
In initializeWecSim (line 157)
In run (line 91)
In wecSim (line 43) 
Elapsed time is 0.418544 seconds.

WEC-Sim Pre-processing ...   
````

To validate the accuracy of the results I’m obtaining in WEC-Sim, I modeled the same WEC in ANSYS/AQWA under identical conditions and properties. However, the results from WEC-Sim differ significantly from those in AQWA. Are there fundamental differences in the assumptions or simplifications made by WEC-Sim compared to hydrodynamic response of AQWA that could explain these discrepancies?

I would appreciate any guidance or insights you could provide on these issues.

MShabara, 2024-09-26:
The pto(i).equilibriumPosition defines the PTO's equilibrium orientation, while pto(i).initial sets its initial orientation. Both parameters can be adjusted to achieve the desired steady-state response, depending on user requirements.

I submitted [PR #1324](https://github.com/WEC-Sim/WEC-Sim/pull/1324) to allow users to modify the center of gravity and center of buoyancy. However, it hasn't been merged into WEC-Sim's development branch yet. In the meantime, you can manually update the bodyClass.m file on your computer using the changes from [this commit](https://github.com/WEC-Sim/WEC-Sim/pull/1324/commits/212942060b40b1478c168a6efad15f57bfdc1db7) to enable specifying a custom center of gravity.

Mohamad-Namaki, 2024-10-07:
Dear @MShabara,

Thank you very much for your detailed explanation. I believe there might be a potential issue related to the functionality described in #1324. Specifically, modifying the center of gravity (CoG) of a non-hydrodynamic body seems to cause an unintended translational movement equivalent to the defined CoG vector. The commit you referenced, which overwrites the CoG for hydrodynamic bodies, appears to produce the same behavior for non-hydro bodies as well. 

Is there a way to assign a new CoG value without inducing this translation? This movement results in inaccurate outputs, as it implies that the CoG of the hydrodynamic body's CAD file must always be at (0,0,0), which is incorrect and introduces significant complications in the simulation.

Additionally, I have encountered an error when setting `body(1).nonlinearHydro = 2` in WEC-Sim. The error does not occur when this parameter is set to 1 or 0. The error message is as follows:

```
Error using wecSim
Derivative of state 'PitchFloat.Structure_PTO.Pitch_PCC_Revolute_Joint.Rz.w' in block 'PitchFloat/Structure PTO/Pitch PCC (Revolute Joint)' at time 0.0 is not finite. The simulation will be stopped. There may be a singularity in the solution. If not, try reducing the step size (either by reducing the fixed step size or by tightening the error tolerances).
```

I attempted to resolve this by reducing the simulation time step (`simu.dt`) by a factor of 10, but the error persists. Could you please provide any insights or suggestions on how to address this issue?

Thank you in advance for your help.

MShabara, 2024-10-07:
@Mohamad-Namaki 

This might be the reason why WEC-Sim did not allow users to modify the c.g in the `wecSimInputFile.m`. It will confuse the users rather than help them. [With great power comes great responsibility](https://en.wikipedia.org/wiki/With_great_power_comes_great_responsibility). 
_________________________________________
1. Modifying the center of gravity (CoG):

To guarantee that you get accurate simulations if the shift in c.g is large is to rerun the hydrodynamic model to make sure that the hydrodynamic coefficients related to the rotational motions are accurate. Also, when you shift the c.g location, the geometry in the .stl file should also be moved to reflect the new c.g location.

Ref to Nathan's response to Issue #1088, the .stl files for visualization requires the origin of the .stl file to align with the center of gravity. Then the SimMechanics animation will shift those .stl files based on the defined center of gravity in the .h5 file (from you hydrodynamic data).  
_________________________________________
2. Non-linear Hydro feature:

When the nonlinear option is turned on, the geometry file (*.stl) (previously only used for visualization purposes in linear simulations) is used as the discretized body surface on which the nonlinear pressure forces are integrated. As in the linear case, the STL mesh origin must be at a body’s center of gravity.

As mentioned in [the documentation](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation), a good STL mesh resolution is required when using the nonlinear buoyancy and Froude-Krylov wave excitation in WEC-Sim. The simulation accuracy will increase with increased surface resolution (i.e. the number of discretized surface panels specified in the *.stl file), but the computation time will also increase. There are many ways to generate an STL file; however, it is important to verify the quality of the mesh before running WEC-Sim simulations with the nonlinear hydro flag turned on.

For more details of the effect of turning on the non-linear hydro calculation refer to [Nathan's response](https://github.com/WEC-Sim/WEC-Sim/issues/1025#issuecomment-1530241660) 

Please let us know if you still have questions.

Mohamad-Namaki, 2024-10-16:
Dear @MShabara,

Thank you very much for your clarification. After further investigating [the error I mentioned earlier](https://github.com/WEC-Sim/WEC-Sim/issues/1315#issuecomment-2396435181) and reviewing the resources you provided, I’ve realized that the issue arises because the wave elevation at time = 0 and location (0,0,0) is equal to the wave amplitude. This results in a large initial hydrodynamic force, which causes the solver to detect a significant derivative of the force at the PTO, leading to the error.

To resolve this, I’ve been utilizing the wave ramp time feature. However, I was wondering if there is an alternative approach that allows for a phase shift at time = 0, so that the wave elevation starts at 0 rather than at the wave amplitude — similar to the phase difference between sine and cosine functions. Would it be possible to implement such a phase shift to avoid this large initial force?

Thank you in advance for any guidance you can provide.

jtgrasb, 2024-10-21:
Hi @Mohamad-Namaki, the wave phase is a private variable which means it cannot be set by the user. Instead, we recommend using the ramp time to start the wave gradually. You can then start your analysis once the ramp time is complete and the wave elevation is at 0. Some other options would be to use a user defined elevation, specify the wave phase seed (if using irregular waves), or modify the source code so that the wave phase can be changed. If there is a good use-case for changing the wave phase, we are open to making a PR that makes it modifiable. 

As for using mean drift and QTFs from ANSYS AQWA, this is not currently supported. We don't have that many AQWA users, so we only have WAMIT mean drift and QTF enabled. This could be fixed by adding some code in the `readAQWA` script to load in the mean drift and QTF coefficients. (Check out `readWAMIT` for an example of how the variables are loaded in). I will add this to our project board, but I expect it may take some time before it is added to the source code. It may be faster for you to make this change if needed soon and make a PR to contribute to WEC-Sim source code.

MShabara, 2024-10-22:
Hi @Mohamad-Namaki 

@jtgrasb provided a thorough response to your comment, but I wanted to add that we would love to collaborate with you on adding AQWA QTFs to WEC-Sim. This would not only enhance your own model but also benefit the wider WEC modeling community by making AQWA QTF integration publicly available.

If you're interested, please send your AQWA QTF results to the email listed in [this link](https://research-hub.nrel.gov/en/persons/mohamed-shabara). I can guide you through the process of modifying BIMIO to read these files and assist with creating a PR for the WEC-Sim repository. This would be a great opportunity to contribute to the open-source community and extend WEC-Sim’s capabilities, which could also increase the visibility of your work.

Mohamad-Namaki, 2024-10-30:
Dear @jtgrasb & @MShabara,

Thank you for your previous guidance. I have modeled this pitch-type Wave Energy Converter in both WEC-Sim and ANSYS AQWA, maintaining identical conditions and properties across both platforms to validate the accuracy of the WEC-Sim results. However, I’ve observed a significant discrepancy between the two sets of results. The outputs from AQWA appear more reasonable, while those from WEC-Sim do not align as expected.

Could there be fundamental differences in the assumptions or simplifications that WEC-Sim applies compared to AQWA’s hydrodynamic response calculations? For additional context, I have disabled QTF effects and other nonlinear aspects within AQWA to ensure a more direct comparison.

I also want to express my enthusiasm for the possibility of collaborating on incorporating AQWA QTFs into WEC-Sim. I believe this integration would not only benefit my own model but also serve the broader WEC modeling community by making AQWA QTF integration publicly available. Although I am currently quite busy, I would be very eager to work with the WEC-Sim development team in a few months when my schedule allows. 

Thank you again for your assistance and consideration.

jtgrasb, 2024-11-01:
Hi @Mohamad-Namaki I personally have not used Ansys AQWA's time-domain solver, but WEC-Sim has been successfully [verified against AQWA](https://asmedigitalcollection.asme.org/OMAE/proceedings-abstract/OMAE2014/45547/279010) in the past. Based on this, I would think there are not fundamentally different assumptions except any optional ones (nonlinear, QTFs, maybe drag?). I would recommend you start with a simple case (one body, no pto, etc.) and make sure all the settings are the same (wave conditions, mass properties, hydrostatic properties, hydrodynamic coefficients, etc.) between WEC-Sim and AQWA, then go step by step to see where the mismatch occurs that matches in WEC-Sim vs. AQWA. If you're not able to identify the cause, please share the differences you are seeing between the results.

MShabara, 2024-11-05:
Hi @ Mohamad-Namaki 

Does both model behave similarly when you try the decay tests? If they do not behave similarly in a simple decay test then the models setup are not the same. I guess the problem you have might be related to how both solvers define the global and body reference frames when the simulation is initialized. 

For the QTFs, can you share your AQWA results files that include the QTF data with us? We will try to implement them in BIMIO and share them with you for validation, we can start with a zero wave wave heading then expand this to multiple wave headings.

jtgrasb, 2024-11-20:
@Mohamad-Namaki Thank you for all your inquiries. Are you able to share AQWA results so we can use them to implement AQWA QTFs in BEMIO? If so, that would be helpful, but no problem if not.

Mohamad-Namaki, 2024-11-24:
Dear @jtgrasb,
Thank you so much for your guidance. I am very interested in collaborating with your team to integrate AQWA QTFs into WEC-Sim. This would be an exciting opportunity to contribute to the development of WEC-Sim while also benefiting the wider WEC modeling community.
Could you kindly share your email address so that I can reach out to discuss this further? I look forward to working together on this project.

MShabara, 2024-11-25:
@Mohamad-Namaki 

Thanks for being open to support WEC-Sim development efforts. You can contact our team via the email in [this link](https://research-hub.nrel.gov/en/persons/mohamed-shabara).

This thread is now long and has multiple topics were discussed, user Defined inputs, device rotations, QTFs. I will close this issue and you can open a new one if you have a new issues or questions if that works for you.

MShabara, 2024-11-27:
@Mohamad-Namaki 

I just want to double check with you about the WEC-Sim version you are using in your model. Is it v4.1 or 6.1? 
Would you be able to test different WEC-Sim versions (6.0 and 6.1.1) and let us know whether your results converges to OrcaFlex results or not?

Mohamad-Namaki, 2024-12-28:
Hi dear @MShabara 
Sorry for my late response. It's WEC-Sim version 6.1 . I will do that and share the results with you. but i am not very familiar with changing the version and I would appreciate if you could provide any links to help me in that matter.

MShabara, 2024-12-30:
@Mohamad-Namaki 

No worries, the `readAQWA.m` file has undergone multiple changes that would take time to detail fully here. These changes stem from updates to the ANSYS AQWA output files as well as bug fixes within the `readAQWA.m` file itself. However, the figure below highlights some of the pull requests (PRs) that contributed to these changes.
 
[image]

jtgrasb, 2025-01-29:
@Mohamad-Namaki I am just following up on this to see if you're still able to collaborate on improving readAQWA.m. It would be very helpful if you could share AQWA results including QTFs. Could you email [Mohamed](https://research-hub.nrel.gov/en/persons/mohamed-shabara) to continue this discussion?

jtgrasb, 2025-02-05:
Closing the issue due to inactivity. @Mohamad-Namaki If you're able to share AQWA results, let us know.

closed 2025-02-05 by jtgrasb