WEC-Sim/WEC-Sim #1323: [Feature request] Precompiled Binary/Porting to any open source compiler?

nedisy, 2024-09-14:
**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.** 

**Is your feature request related to a problem? Please describe.**
_Is it really open source if it's run on on Matlab?_

**Describe the solution you'd like**
_Precompoled Binary that can run without any Matlab hassle would be great._

**Describe alternatives you've considered**
_No (truly) open source alternative found till today._

**Personal project context**
_I just want to stimulate supported (moor?) floating body and got the supports' force._

**Additional context**
_._

MShabara, 2024-09-17:
Hi @nedisy,

Thank you for your questions. Here are the answers:
1. > Is it really open source if it's run on Matlab?
   
Yes, it can be considered open-source because you can access and modify the WEC-Sim source code. However, it is not free, as it requires multiple Matlab licenses to use.

2. > A pre-compiled binary that eliminates Matlab requirements would be a great addition. 

The WEC-Sim team will look into this and get back to you.

3. >  No (truly) open source alternative found till today.
   
For a free-open source WEC modeling software you can check [HydroChrono](https://github.com/NREL/HydroChrono). However, it is still in version 0, with version 1 yet to be released next year. Only a few free decay tests have been validated, so if you create a model with version 0 in waves, you will need to validate it yourself. Nothing that version 0 does not support advanced mooring systems, but you can use simple spring/damper elements to account for the mooring effects. 

4. > I just want to simulate a supported (moored) floating body and retrieve the support forces.

WEC-Sim can be coupled with MoorDyn for mooring simulations. Check out the [WEC-Sim Applications](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/Mooring) repository for examples of WEC-Sim and MoorDyn coupling.

dforbush2, 2024-09-18:
@MShabara covered this well, I wanted to note that you can also run MoorDyn in a stand-alone way, without coupling to WEC-Sim. With your modeling intent that might be sufficient.