WEC-Sim/WEC-Sim #1329: [BUG] Irregular waves PM spectrum generation in parallel

dododododo21, 2024-09-19:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
When I generate irregular waves with the PM spectrum with a given phaseSeed (in my case = 2), I get two different wave elevations if I am running wecSim and wecSimPCT.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): R2022b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.0

**WEC-Sim file(s):**
I don't know.

**How to reproduce:**
I tried to write a very simple code, which should keep all the constants as I described them in my InputFile:
clear all 
close all

%% run normal
wecSim

figure
hold on;
phaseSeed = waves.phaseSeed;
plot(waves.waveAmpTime(:,1),waves.waveAmpTime(:,2), 'DisplayName', ['Normal phaseSeed=', num2str(phaseSeed)])

%% create .mat file
mcr.header = ["waves.height", "waves.period"];

mcr.cases(1 , 1) = waves.height;
mcr.cases(1, 2) = waves.period;

save MCR_cases.mat mcr

%% run in parallel
wecSimPCT
load('savedData001.mat')
phaseSeed = waves.phaseSeed;
plot(waves.waveAmpTime(:,1),waves.waveAmpTime(:,2), 'DisplayName', ['Parallel phaseSeed=', num2str(phaseSeed)])
xlabel('time [s]')
ylabel('Generated waves [m]')
legend

**Expected behavior**
I expect the same waves.

**Screenshots:**
[image]

**Additional information:**
Maybe I made another mistake or it's already resolved in the new version, but after a day of figuring out my differences of the results I calculated, I thought it was relevant to post.

dforbush2, 2024-09-23:
Hi @dododododo21 

Can you please confirm that if you run wecSim multiple times with phaseSeed =2, and if you run wecSimPCT with phaseSeed = 2, each of those runs are identical? In other words, the first run of wecSim is equal to the second run of wecSim, and the first run of wecSim PCT is equal to the second run of wecSimPCT?

dododododo21, 2024-09-25:
Yes they are the same. This is the result when wecSimPCT is called twice like (the rest of the code is the same as above):
for i_run = 1:2
    wecSimPCT
    load('savedData001.mat')
    phaseSeed = waves.phaseSeed;
    plot(waves.waveAmpTime(:,1),waves.waveAmpTime(:,2), 'DisplayName', ['Run', num2str(i_run),': Parallel phaseSeed=', num2str(phaseSeed)])
end

[image]
(Run2 completely overlays Run1)

akeeste, 2024-10-30:
It looks like [wecSimPCT ](https://github.com/WEC-Sim/WEC-Sim/blob/8177e1cf593616c3ec4854d81d319adb402897f8/source/wecSimPCT.m#L65-L74)manipulates the phaseSeed in some way;

Not sure if this is causing the issue, but is a starting place. I think it is worthwhile fixing these cases to be the same as the phase seed's exact purpose is to create identical irregular waves and is not in this case.

kmruehl, 2024-11-12:
@dododododo21 thanks for bringing this to our attention. I was able to reproduce your error and I'm looking into a resolution.

kmruehl, 2024-11-12:
I first attempted to force `phaseSeed =1` for the the MCR case, shown below. That didn't resolve the issue. 
```matlab
mcr.header = ["waves.height", "waves.period", "waves.phaseSeed"];
mcr.cases(1 , 1) = waves.height;
mcr.cases(1, 2) = waves.period;
mcr.cases(1, 3) = waves.phaseSeed;
save MCR_cases_phase.mat mcr
```

I also commented out lines 65-74 in `wecSimPCT`. That didn't resolve the issue. 
> It looks like [wecSimPCT ](https://github.com/WEC-Sim/WEC-Sim/blob/8177e1cf593616c3ec4854d81d319adb402897f8/source/wecSimPCT.m#L65-L74)manipulates the phaseSeed in some way;
> 
> Not sure if this is causing the issue, but is a starting place. I think it is worthwhile fixing these cases to be the same as the phase seed's exact purpose is to create identical irregular waves and is not in this case.
> 
> ```
>     numConditions=2;
>     if length(waves.phaseSeed)>1
>         numConditions=numConditions+1;
>         mcr.header{numConditions} = 'waves.phaseSeed';
>         len = length(mcr.cases(:,1));
>         for nseed=1:length(waves.phaseSeed)
>             mcr.cases(len*(nseed-1)+1:len*(nseed-1)+len,1:numConditions-1) = mcr.cases(1:len,1:numConditions-1);
>             mcr.cases(len*(nseed-1)+1:len*(nseed-1)+len,    numConditions) = waves.phaseSeed(nseed);
>         end
>     end
> ```

I'll dig into the WEC-Sim source a bit more, but this may actually have to do with the way parallel commuting works. I wonder if the same issue exists with `wecSimMCR`, I suspect not.

kmruehl, 2024-11-12:
@dododododo21 I was able to confirm that this issue only exists for `wecSimPCT`, and not for `wecSimMCR`. I recommend using `wecSimMCR` for now. I suspect that this has to do with the way parallel commuting works, and may not be an issue with the WEC-Sim source. 

Here's my script to run `wecSim` and `wecSimMCR`, both with `waves.phaseSeed=1`

```matlab
clear all
close all

%% run wecSim
wecSim

figure
hold on;
phaseSeed = waves.phaseSeed;
plot(waves.waveAmpTime(:,1),waves.waveAmpTime(:,2), 'DisplayName', ['Normal phaseSeed=', num2str(phaseSeed)])

%% create .mat file
mcr.header = ["waves.height", "waves.period", "waves.phaseSeed"];
mcr.cases(1 , 1) = waves.height;
mcr.cases(1, 2) = waves.period;
mcr.cases(1, 3) = waves.phaseSeed;
save MCR_cases_phase.mat mcr

%% run MCR
wecSimMCR

phaseSeed = waves.phaseSeed;
plot(waves.waveAmpTime(:,1),waves.waveAmpTime(:,2), 'DisplayName', ['MCR phaseSeed=', num2str(phaseSeed)])
xlabel('time [s]')
ylabel('Generated waves [m]')
legend
```
And here are my results comparing the two runs:
[image]

kmruehl, 2024-11-12:
Note to self, here's something to look into: https://www.mathworks.com/matlabcentral/answers/787904-default-random-seed-when-parallel-pool-is-created

jleonqu, 2025-05-15:
This issue was solved in PR #1467