WEC-Sim/WEC-Sim #1339: [BUG]

tkabrekken, 2024-10-09:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
WEC-Sim tutorial page for the RM3 states the body masses are 727 and 827 tons.  But the RM3 body classes define the masses as 4.7979e6 and 2.1918e7.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): [e.g., 2022b, 2023a, 2023b, 2024a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): [e.g., v5.1, v6.0, v6.1]

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
_How can we reproduce the error? Describe the steps to reproduce the error._

**Expected behavior**
_What is the excepeted behavior? Describe the expected behavior._

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screenshot of the error message from the MATLAB command window._

**Additional information:**
_Provide any other additional information._

jtgrasb, 2024-10-10:
Hi @tkabrekken, I am not getting the same results you are. The masses are set as 'equilibrium' in the wecSimInputFile, which sets the mass equal to the displaced volume times the water density. When I run the RM3 example case from the WEC-Sim main branch, I get 886687 kg for the spar and 725833 kg for the float. Please check if there is anything you have modified in the example.

kmruehl, 2024-10-30:
**10/30 Discussion**
There is no reason for us to update `body.mass`, we typically use `hydroForce` for WEC-Sim calculations. @akeeste will look into changing this workflow in conjunction with variable hydro.