WEC-Sim/WEC-Sim #1346: [BUG] Calculation differences between v6.1 and v6.0

hachikoi1, 2024-10-27:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
Hi developers,

Recently I update my WEC-Sim from version 6.0 to version 6.1. But I found the two version have significant caculation differences for my multi-body case. Here is my model.

[image]

I used the same .h5 file, and the same simulation settings, with only the difference in WEC-Sim version.

(1) To further troubleshoot the bug, I did not use MoorDyn, but the problem still exists. 

(2) I try to replace `v6.1\source\objects\ptoClass.m` and `v6.1\source\lib\WEC-Sim\WECSim_Lib_PTOs.slx` with `v6.0\source\objects\ptoClass.m` and `v6.0\source\lib\WEC-Sim\WECSim_Lib_PTOs.slx`, but the issue still exists. So it doesn't seem to have anything to do with the PTO.

[image]

(3) Then I presume the issue is related to the `body` module. Since version 6.1 add some advanced function, I can't pinpoint what went wrong. Perhaps you could you can quickly identify the location of the error.

I upload my case [here](https://github.com/user-attachments/files/17532381/Test6.0.zip). Looking forward to you guys fixing this.

Best regards,

Tianyuan

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): [e.g., 2022b, 2023a, 2023b, 2024a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): [e.g., v5.1, v6.0, v6.1]

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
_How can we reproduce the error? Describe the steps to reproduce the error._

**Expected behavior**
_What is the excepeted behavior? Describe the expected behavior._

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screenshot of the error message from the MATLAB command window._

**Additional information:**
_Provide any other additional information._

hachikoi1, 2024-10-28:
Above results are obtained by setting `pto(1:3).damping =  0;`

When I specify a pto damping 3e6 for the simulation, the oscillations in accelerations disappear, but the calculation differences still exist.

[image]

dforbush2, 2024-10-28:
I'll look into this. thank you for providing your input files. Your observations above already ruled out any most of the first things I would try...so I will follow up when I have something. 

In the meantime I do recommend you pull down our bugfix release v6.1.1.

hachikoi1, 2024-10-28:
> I'll look into this. thank you for providing your input files. Your observations above already ruled out any most of the first things I would try...so I will follow up when I have something.
> 
> In the meantime I do recommend you pull down our bugfix release v6.1.1.

Dear [dforbush2](https://github.com/dforbush2)

Sorry for my confusion. The WEC-Sim version I used is the lastest. I download it on Saturday (2024.10.26), so it should be the version 6.1.1.

Kind regards,

Tianyuan

dforbush2, 2024-10-28:
Can you please provide a bit more detail on how you have created figure 2 in your first post, with the spiky acceleration? I am unable to reproduce this output. Also, which MATLAB version are you using?

hachikoi1, 2024-10-28:
Hi [dforbush2](https://github.com/dforbush2)

Figure 2 is the comparisons of Body1 heave displacement between version 6.0 and 6.1.1. And the line of `version 6.1 restore PTO` (It should be version 6.1.1 restore PTO) is obtained by replacing the PTO module `v6.1\source\objects\ptoClass.m` and `v6.1\source\lib\WEC-Sim\WECSim_Lib_PTOs.slx` with `v6.0\source\objects\ptoClass.m` and `v6.0\source\lib\WEC-Sim\WECSim_Lib_PTOs.slx`.

For version 6.1.1, I add the WEC-SIM v6.1.1 source path into Matlab environment and conduct simulations. Repeat this step to get the results for version 6.0.

hachikoi1, 2024-10-28:
As for the acceleration, you can run `compareResults.m` to obtain it.

```
clear all; close all; clc;
load('./Data/v60.mat');
v60Disp = output.bodies(1).position(:,3);
v60Acc = output.bodies(1).acceleration(:,3);
clear output

load('./Data/v61.mat');
v61Disp = output.bodies(1).position(:,3);
v61Acc = output.bodies(1).acceleration(:,3);
figure;
hold on
plot(output.bodies(1).time,v60Acc,'k-');
plot(output.bodies(1).time,v61Acc,'b-.');
legend({'v6.0','v6.1'});
hold off
```

dforbush2, 2024-10-29:
Hmm I continue to be unable to create any significant difference between release results or the spiky acceleration plot. I have tried your provided input file in both v6.0 and the most recent code base. 

Can you please clarify which MATLAB version you are using? I have been testing on 2023b.

Please also confirm your input file parameters
%% Simulation Data
simu                            =  simulationClass();
simu.simMechanicsFile           =  'model.slx';
simu.b2b                        =  1;
simu.solver                     =  'ode4';
simu.explorer                   =  'off';
simu.startTime                  =  0;
simu.rampTime                   =  0;
simu.endTime                    =  200;
simu.dt                         =  0.0125; % simu.dtOut = simu.dt
simu.gravity                    =  9.80665; % Default = 9.81
simu.rho                        =  1025; % Default = 1000
simu.stateSpace                 =  0; % 0 (convolution integral), 1 (state-space). Default = 0
simu.cicDt                      =  simu.dt; % Default = 'dt'
simu.cicEndTime                 =  60; % Default = 60, recommended by OpenFAST
simu.explorer                   =  'off';  

%% Wave Information
waves                           =  waveClass('irregular');
waves.bem.range                 =  0.1:0.05:3.5; % Use shared frequency range between WAMIT and AQWA
waves.height                    =  5.5253;
waves.period                    =  11.1546;
waves.spectrumType              =  'JS';
waves.phaseSeed                 =  1; % Default = 0
waves.direction                 =  0.0; % Default = 0

%% Body Data
body(1)                         =  bodyClass(['./hydroData/model.h5']);
body(1).mass                    =  'equilibrium';
body(1).inertia                 =  [1.2507E+10 1.2507E+10 2.3667E+10];

body(2)                         =  bodyClass(['./hydroData/model.h5']);
body(2).mass                    =  body(1).mass;
body(2).inertia                 =  [1.3312e+08 1.3312e+08 2.1384e+08];

body(3)                         =  bodyClass(['./hydroData/model.h5']);
body(3).mass                    =  body(1).mass;
body(3).inertia                 =  body(2).inertia;

body(4)                         =  bodyClass(['./hydroData/model.h5']);
body(4).mass                    =  body(1).mass;
body(4).inertia                 =  body(2).inertia;

%% PTO and Constraint Parametersre
constraint(1)                   =  constraintClass('Constraint1');
constraint(1).location          =  [0 0 0];
constraint(1).initial.displacement =  [0 0 0]; 

pto(1)                          =  ptoClass('PTO1');
pto(1).location                 =  [-51.75 0 0]; % Location in the global coordinate system
pto(1).hardStops                =  struct('upperLimitSpecify',               'on',...
    'upperLimitBound',                 15-(12.5-12.5/2), ... % PTO does not have a concept of WEC height, which is added here
    'upperLimitStiffness',             1e10, ...
    'upperLimitDamping',               1e10, ...
    'upperLimitTransitionRegionWidth', 1, ...
    'lowerLimitSpecify',               'on',...
    'lowerLimitBound',                 -13+12.5/2, ... % PTO does not have a concept of WEC height, which is added here
    'lowerLimitStiffness',             1e10, ...
    'lowerLimitDamping',               1e10, ...
    'lowerLimitTransitionRegionWidth', 1);

pto(2)                          =  ptoClass('PTO2');
pto(2).location                 =  [51.75*cos(pi/3) -51.75*sin(pi/3) 0]; % Location in the global coordinate system
pto(2).hardStops                =  pto(1).hardStops;

pto(3)                          =  ptoClass('PTO3');
pto(3).location                 =  [51.75*cos(pi/3) 51.75*sin(pi/3) 0]; % Location in the global coordinate system
pto(3).hardStops                =  pto(1).hardStops;

hachikoi1, 2024-10-29:
Hi [dforbush2](https://github.com/dforbush2)

I also use Matlab 2023b. I tried it again by downloading the WEC-Sim from repository, and the spiky acceleration still exist. Below is my step.

The steps are as follow.

1. Download WEC-Sim-main.zip from https://github.com/WEC-Sim/WEC-Sim and decompress it.
2. Add the source path to Matlab environment.
3. Run the case file
4. plot(output.bodies(1).acceleration(:,3)), then the figure is
[image]

Did you download your WEC-Sim from the repository? Was the source files of the old version removed from the Matlab environment? I tried another desktop and spiky acceleration happened as well.

dforbush2, 2024-10-29:
I have it now. It differs between some of our forks and branches so I'm close to narrowing it down. I'm guessing it is an improperly linked library.

dforbush2, 2024-10-29:
Update for future me tomorrow: 
* generated body classes are identical except for the hf1 substructure for variable hydro
* b2b ON: the v611 has spiky acceleration. b2b OFF: the v60 has spiky acceleration. 
* excitation Force is identical. 
* signal routing identical in both cases

dforbush2, 2024-10-29:
Some progress:

those acceleration spikes are the result of the PTOs hitting their end stops, where a lot of force is suddenly applied over a small displacement. 

Those end stops also exist in v6.0...but the kinematics of the PTO in this simulation are markedly different so they do not approach their end stops. I will continue to explore why these kinematics differ so significantly.

dforbush2, 2024-10-29:
I FOUND IT!
Improper library link in the Goto/From flags introduced with the variable hydro feature merge (v6.1). Bodies 2-4 are erroneously using the IRKB from body 1. in v60 this error does not exist because there is no variable hydrofeature, and these values are read from the top level body mask. I will introduce a PR to fix this problem.

In the mean time: you can trust your results from v6.0. Thank you for finding this bug.

hachikoi1, 2024-10-29:
> I FOUND IT! Improper library link in the Goto/From flags introduced with the variable hydro feature merge (v6.1). Bodies 2-4 are erroneously using the IRKB from body 1. in v60 this error does not exist because there is no variable hydrofeature, and these values are read from the top level body mask. I will introduce a PR to fix this problem.
> 
> In the mean time: you can trust your results from v6.0. Thank you for finding this bug. I will leave this issue open until resolved with a PR and credit @hachikoi1 in the referencing PR.
> 
> @akeeste I may have inadvertently introduced this bug during my merge of this PR. However, I'm not sure it would have been caught by the single-body unit test. I believe the problem needs to be addressed in the Hydrodynamic Body block mask. I will follow up.

Congratulations! The problem seems to involve the underlayer development. I really don't have a clue about this without your help.

May I know your name. I'll keep you in mind.

Best regards,

Tianyuan

akeeste, 2024-10-30:
@dforbush2 agreed. It looks like the Hydro Body mask is not updating the From flag in the radiation convolution block. I'm surprised that the RM3 regression test did not catch this as it checks the position time series of bodies 1-2 pretty closely

commit 1cdd4631 (2024-10-30): Body block library update for bug fix #1346  (#1347)

* fix on pDis function call

* preliminary fix for issue #1288

* Update readAQWA.m (#1253)

* body mask fixes for #1346

* add warning when variable hydro turned on for drag or nonhydro bodies

* flex body - add missing mask lines, define h5 button callback

---------
commit 59b938a0 (2024-11-04): WEC-Sim QTFs compatability with NEMOH (#1353)

* Fixing documentation inconsistency for PTO-Sim blocks (#1330)

* Fixing documentation inconsistency for PTO-Sim blocks

* Adding comments to nondimensional values for clarification

* update analytics implementation in docs re deprecation of analytics_id (#1342)

* Body block library update for bug fix #1346  (#1347)

* fix on pDis function call

* preliminary fix for issue #1288

* Update readAQWA.m (#1253)

* body mask fixes for #1346

* add warning when variable hydro turned on for drag or nonhydro bodies

* flex body - add missing mask lines [... cut]
closed 2024-10-30 by dforbush2