WEC-Sim/WEC-Sim #1356: `waveClass` amplitude definition for irregular waves

mmacn42, 2024-11-04:
**Description of Bug**
In `waveClass` > `irregWaveSpectrum` I'm wondering if `obj.amplitude` is calculated incorrectly. From line 708 I see the following code:

```
                    obj.spectrum = obj.spectrum(wn(2:end-1));                             % Wave Spectrum [m^2-s/rad] 
            end
            obj.amplitude = 2 * obj.spectrum;                                              % Wave Amplitude [m]
```
This is not the definition of amplitude I am used to seeing, nor do the units agree.

**Workstation information:**
 - Operating System: Windows 11 Pro
 - MATLAB/Simulink Release: 2024a
 - WEC-Sim Release: 6.1.1, working off of commit "1cdd4631"

**WEC-Sim file(s):**
The definition is in `waveClass`. Not sure where else `waves.amplitude` is used 

**How to reproduce:**
A wave with the following inputs produces the `spectrum` and `amplitudes` in the plot below: 
```
waves = 

  waveClass with properties:

               bem: [1×1 struct]
           current: [1×1 struct]
         direction: 180
     elevationFile: 'NOT DEFINED'
             gamma: 1.5526
            height: 1
            marker: [1×1 struct]
            period: 3
         phaseSeed: 0
      spectrumFile: 'NOT DEFINED'
      spectrumType: 'JS'
               viz: [1×1 struct]
        waterDepth: 69.8580
            spread: 1
         amplitude: [500×1 double]
         deepWater: []
            dOmega: [500×1 double]
             omega: [500×1 double]
             phase: [500×1 double]
             power: 1.2816e+03
          spectrum: [500×1 double]
              type: 'irregular'
           typeNum: 20
       waveAmpTime: [2001×2 double]
    waveAmpTimeViz: []
        wavenumber: [500×1 double]
```
[image]

**Expected behavior**
I would expect something more like $\frac{1}{2}A^2(\omega)=S(\omega)\Delta\omega$, which gives the following:
[image]

MShabara, 2024-11-11:
Hi @mmacn42 ,

That’s a great question! I hadn’t looked closely at this part of the code before.

It seems that the variable name and comment on line 708 could be clarified and the units is not correct. 
The wave amplitude follows the formulation found in several references, including [this one](https://www.sciencedirect.com/science/article/pii/S1364032122009510).

If you substitute the equation in line `708` into `738`, you’ll find they correspond closely to the equation in section (4.3) of the reference mentioned above. Additionally, lines 739 and 740 are quite similar to Eq. (112) in the same reference.

Hope this helps! Let me know if you have further questions.

kmruehl, 2024-11-12:
@mmacn42 thanks for pointing out this discrepancy. https://github.com/WEC-Sim/WEC-Sim/blob/6f4be976c971a62637b0eab91c63a2d40e914d0c/source/objects/waveClass.m#L61

lists `waves.amplitude` as 2*(wave spectrum vector) for irregular waves. However the in-line comment is incorrect. I'll submit a PR to resolve the discrepancy.

kmruehl, 2024-11-12:
@MShabara https://github.com/WEC-Sim/WEC-Sim/pull/1366 should resolve this issue. I've requested your review

kmruehl, 2024-11-13:
Resolved with https://github.com/WEC-Sim/WEC-Sim/pull/1366

closed 2024-11-13 by kmruehl