WEC-Sim/WEC-Sim #1361: [BUG] Error running test example for WEC-Sim installation

SPYROHAWK, 2024-11-07:
**Describe the bug (REQUIRED)**
Hello! I apologize if this is not super descriptive or if this is user error, but I have been trying to find a solution for this and I don't think this is on my end. I'm a university student trying to use WEC-Sim and going through the initial install process as outlined in the third tutorial video "3 WEC Sim Installation". Everything is fine up through running the bemio.m file in the examples/RM3/hydroData folder. I then change the path back to the examples/RM3 folder using the cd command, but then when I run the "wecSim" command I am getting an error.

[image]

If my reading of this error is correct, the "wecSim" command runs a "wecSim.m" file. Line 40 of that file has `run('initializeWecSim');` running the "initializeWecSim.m" file. Line 62 of that file has `    run('wecSimInputFile');` which runs the "wecSimInputFile.m" file. This is where my issue of the error seems to fall apart. There don't seem to be any errors occurring from the lines of code within the "wecSimInputFile.m" file, but rather for some reason the `run` command doesn't seem to want to run the file, despite the previous run commands being able to run their files until the error.

PART 2:

After adding the version MatLab 2024b (in case WecSim was just incompatible with the older version) and repeating all of the steps, am getting stuck at the same spot with a new issue. When I run the bemio file I get a lot of warnings about "Colon operands must be real scalars" but it still runs. Then I switch to the examples/RM3 directory and run the wecSim command I get this error:
[image]
I didn't think it would, but for the sake of trying, I did run the bemio file again and re-did the cd command and the wecSim command, same error message.

So I guess this is technically now two different errors: one on the 2020a version and one on the 2024b version, but at the exact same spot. Not sure if they are related or not.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Originally MacOS Sonoma Version 14.6.1 (23G93), in the process of trying to figure out this issue I updated to MacOS Sequoia Version 15.1 (24B83).
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): Originally 2020a, in the process of trying to figure out this issue I also installed 2024b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): 6.1.1

**WEC-Sim file(s):**
I'm not sure exactly, but I put my best guess as to the origin of the problem in the description section.

**How to reproduce:**
Install a new instance of WEC-Sim and try to launch wecSim for the RM3 example.

**Expected behavior**
According to the "3 WEC Sim Installation" video from the WEC-Sim youtube channel, "the RM3 example should run, showing mechanic explorer window, and figures upon completion". Or... basically running as intended instead of failing and giving the error message.

**Screenshots:**
See bug description.

**Additional information:**
I don't have much else to say. I'm trying to use Wec-Sim for a big project, and I keep hitting this roadblock. I'm frustrated, just as anyone is when something isn't working. But I am hoping I made my issue clear enough that it can help get it fixed.

MShabara, 2024-11-11:
Hi @SPYROHAWK 

Thanks for your message and your interest in using WEC-Sim!

1. **For part (1):** 
The error you’re seeing appears to be due to an incorrect input related to the visualization library. Could you confirm that the explorer setting in the `wecSimInputFile.m` is specified as a character? Here’s an example of the correct syntax:

`simu.explorer = 'on';  % Turn SimMechanics Explorer (on/off)`

2. **For part (2):**
Currently, WEC-Sim is not compatible with MATLAB either versions (2020a and 2024b). The supported versions are R2022a, R2022b, R2023a, and R2023b, as indicated in the [workflow file](https://github.com/WEC-Sim/WEC-Sim/blob/main/.github/workflows/run-tests-dev.yml).

Let us know if there’s anything else we can assist you with!

kmruehl, 2024-11-12:
@SPYROHAWK for 2) it looks like the `*.h5` did not properly download, so you'll need to rerun BEMIO for the RM3 example. You'll need to `cd` into `examples/RM3/hydroData` to run `bemio`, that should generate a new `rm3.h5` file. Then you can `cd` back to `examples/RM3` and run `wecSim`. 

Also, the "Colon operands must be real scalars" warnings have been resolved in the dev branch of WEC-Sim with, but that fix is not in the v6.1.1 release. However, they are just warnings about a future deprecation, and they will not affect the bemio run.

SPYROHAWK, 2024-11-22:
> Hi @SPYROHAWK
>
> Thanks for your message and your interest in using WEC-Sim!
>
> 1. **For part (1):**
>    The error you’re seeing appears to be due to an incorrect input related to the visualization [...]

Sorry, just following up on this. This is more of a general MATLAB issue than a WECSIM issue but I figured I'd put it here in case you had any thoughts.

I downloaded R2023b and tried to run it, but it keeps saying I need a java runtime environment (which I have, after all R2024b works fine), so I clearly something else is going on. After looking through it, it seems that none of MATLAB R2022a-R2023b are compatible with Mac Sequoia 15 (which I updated to in the initial attempt to deal with the issue I was having), only R2024 is.

Furthermore, it seems it's not possible to downgrade to my MacOS version without completely wiping my computer.

That all being said, do. you have any suggestion on how I could possibly get WEC-Sim to run on my machine?

kmruehl, 2024-11-27:
@SPYROHAWK sorry, it sounds like an issue with MATLAB. I recommend contacting MathWorks support and see what they recommend. I'm going to go ahead and close this issue, but feel free to reopen it once you have MATLAB running and we can help with WEC-Sim.

closed 2024-11-27 by kmruehl