WEC-Sim/WEC-Sim #1363: Inconsistent motions with multiple waveClasses

mmacn42, 2024-11-08:
**Describe the bug**
I'm trying to use multiple waveClasses to simulate multi-spectral ocean conditions and seeing unrealistic WEC behaviors. To try and better understand, I ran this simple comparison between two test cases: 

_Test 1_
WEC-Sim RM3 example model
A single `regularCIC` wave, with height 2.0 meters and period 13 seconds

_Test 2_
WEC-Sim RM3 example model
Four waveClasses, each with `regularCIC` waves and one quarter of the wave height of _Test 1_ (0.5 meters). Same period.

**Workstation information (REQUIRED):**
Describe the workstation:
- Operating System: Windows 11 Pro
- MATLAB/Simulink Release: 2024a
- WEC-Sim Release: 6.1.1, working off of commit "1cdd4631"

**WEC-Sim file(s):**
I've uploaded the `wecSimInputFile` for each test case and the `userDefinedFunctions` I used to generate the plots I pasted below. The model I am running in this test case is the RM3 in the WEC-Sim examples folder.

[multiwavetests.zip](https://github.com/user-attachments/files/17683589/multiwavetests.zip)

**How to reproduce:**
Running each of the two `wecSimInputFile`s should reproduce the models and plots I've included.

**Expected behavior**
I expect the two models to have identical responses, as the superimposed waves are the same. However I see very different responses and forces on both bodies, shown below. I wonder if something is being double counted? Or if there is a mistake in my implementation please let me know.

**Screenshots:**
The two tests produce identical wave elevations but different body motions and forces.

_Wave elevations:_
[image]

_Body motions:_
[image]
[image]

_Body forces:_
[image]
[image]

MShabara, 2024-11-12:
Hi @mmacn42,

The input files you attached are identical. Identical input files will produce identical results (assuming you used same .h5 and same .slx files).

I tested both input files you provided, using the geometry, hydroData, and .slx file from the examples folder. After running both input files, I found that they generated identical results. Unfortunately, I wasn’t able to replicate the issue you described.

Let me know if there’s anything specific you’d like me to check or if there are additional details that could help troubleshoot this further.

Best,

mmacn42, 2024-11-12:
Very sorry, I attached the wrong Test 2 file. Here is the correct one.

[wecSimInputFile_test2.zip](https://github.com/user-attachments/files/17719199/wecSimInputFile_test2.zip)

MShabara, 2024-11-12:
Hi @mmacn42 

The current version of WEC-Sim does not define irregular waves by using multiple regular wave instances. While I understand that this approach can work in other software (such as OpenFOAM v11), it is not applicable in WEC-Sim. To define irregular waves, you'll need to use the specific options available in the WEC-Sim documentation for creating a single irregular wave instance.