WEC-Sim/WEC-Sim #1367: [BUG] WEC-Sim v6.0 is not equivalent to v6.1.1 for OSWEC

jtgrasb, 2024-11-12:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
When running OSWEC example using v6.1.1 in irregular waves, the results are different from v6.0. The regression tests did not seem to catch this because they only run on the RM3.

**Workstation information (REQUIRED):**
 - Operating System: Windows 11
 - MATLAB R2023b
 - WEC-Sim v6.0 vs. v6.1.1

**WEC-Sim file(s):**
It seems likely that it is something in either the bodyClass or body library, but I have not diagnosed the issue yet. 

**How to reproduce:**
Run the OSWEC example using v6.0 and compare to results from v6.1.1

**Expected behavior**
v6.0 should yield the same results as v6.1 for the OSWEC example to my knowledge. Note below that the difference occurs in irregular waves, but not regular waves.

**Screenshots:**
Irregular Waves:
v6.1.1:
[image]
v6.0:
[image]
In regular waves: 
v6.1.1:
[image]
v6.0
[image]

**Additional information:**
@MShabara @akeeste Any idea where these differences would be coming from?

akeeste, 2024-11-13:
@jtgrasb Nothing immediate stands out, though we made a lot of changes recently. Can you plot the pitch position of the two cases together and the pitch forces of the two cases together so that we can assess the differences more easily?

dforbush2, 2024-11-13:
contrary to #1346 and #1350, in this case the excitation force IS different. Strongly recommend starting there with this: I suspect the other force differences may follow from this initial difference.

dforbush2, 2024-11-13:
good find @jtgrasb 
to confirm, the change introduced to #1289 is the correct handling of waveSpread and the reference case should be updated.

closed 2024-11-29 by jtgrasb