WEC-Sim/WEC-Sim #1368: ptoInternalMechanics incorrect when a PTO has non-zero pretension or equilibrium position

akeeste, 2024-11-13:
**Describe the bug**
The PTO library "translational PTO" calculates ptoInternalMechanics incorrectly when a PTO has non-zero pretension or equilibrium position. Other blocks may do this too. The equilibrium position is not taken into account when calculating the spring force 

**Workstation information (REQUIRED):**
 - Operating System: Windows 
 - MATLAB 2024a
 - WEC-Sim v6.1

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
In the RM3 example, to the ``wecSimInputFile.m`` add:
```
%% Initial displacement
pto(1).stiffness = 120000;                           % PTO Stiffness [N/m]

% Option 1
body(1).initial.displacement = [0 0 0.1];
body(2).initial.displacement = [0 0 0];
dz = body(1).initial.displacement(3) - body(2).initial.displacement(3);
pto(1).pretension = pto(1).stiffness*dz;

% % Option 2
% pto(1).extension.PositionTargetSpecify = 1;
% pto(1).extension.PositionTargetValue = 0.1;
```

to the ``userDefinedFunctions.m`` add:
```
% Decay test
figure()
plot(output.bodies(1).time, output.bodies(1).position(:,3)-body(1).centerGravity(3), ...
     output.bodies(2).time, output.bodies(2).position(:,3)-body(2).centerGravity(3));
xlabel('Time (s)');
ylabel('Heave position (m)');
legend('Float','Spar');
xlim([0 10]);

fi = output.ptos(1).forceTotal - output.ptos(1).forceConstraint - output.ptos(1).forceActuation;
figure()
plot(output.ptos(1).time, output.ptos(1).powerInternalMechanics(:,3),...
     output.ptos(1).time, output.ptos(1).velocity(:,3).*(-(output.ptos(1).position(:,3)-0)*pto(1).stiffness - output.ptos(1).velocity(:,3)*pto(1).damping), '--',...
     output.ptos(1).time, output.ptos(1).velocity(:,3).*fi(:,3), ...
     output.ptos(1).time, output.ptos(1).velocity(:,3).*(-(output.ptos(1).position(:,3)-pto(1).equilibriumPosition)*pto(1).stiffness - output.ptos(1).velocity(:,3)*pto(1).damping), '--');
xlabel('Time (s)');
ylabel('PTO power (W)');
title('PTO power (W)');
xlim([0 10]);
legend('PowerInternalMechanics','v*(-k*x - c*v)','v*(-k*(x-x0) - c*v)','v * (f_t - f_c - f_a)');
```

**Expected behavior**
The forceInternalMechanics (force due to the joints stiffness and damping) should be equivalent to the joint's forceTotal - forceActuation - forceConstraint. The joint automatically calculates the latter values and better serve as truth of what forces are applied during the simulation.

**Screenshots:**
[image]
[image]

**Additional information:**
I need to test one more case that includes an actuation force along the joint to confirm that:
F_total = F_constraint + F_internal + F_actuation

akeeste, 2024-11-13:
Further detail-
This bug does not appear when using the PTO extension feature instead of initial displacement because the change in position is inherently built into the PTO position. And I confirmed in the below plots that F_actuation is appropriately account for in the above:

With force actuation and PTO initial displacement:
[image]

with force actuation and PTO extension:
[image]

akeeste, 2024-11-13:
I'll put in a PR today for this. One outstanding question: is it really accurate to be calculating the power from internal mechanics using the combined spring and damping forces? The damper is what is dissipating power from the system, whereas the spring is storing energy and returning it to the system. I think this could be misleading when users think about the power they are absorbing/removing from the system. Perhaps we can retain the ``forceInternalMechanics`` but remove ``powerInternalMechanics``

akeeste, 2024-11-13:
@MShabara I see you assigned yourself to this PR so I tagged you as reviewer in #1369. This fix is necessary for our TEAMER modeling

MShabara, 2024-11-13:
Hi @akeeste 

A few months ago, I noticed that the constraint force calculation for models with initial displacement differs from the results we get in Hydrochrono, even though the device response was exactly the same. I haven’t had the chance to investigate the cause of this discrepancy yet, I assigned myself because I was hoping that this can fix this discrepancy problem.

I agree with your observation. As per the MATLAB documentation, the total force includes actuation, internal, limit, and constraint forces. So calculating the internal force based on your equation seems more rigorous.

closed 2024-11-13 by MShabara

akeeste, 2024-11-13:
> Hi @akeeste
>
> A few months ago, I noticed that the constraint force calculation for models with initial displacement differs from the results we get in Hydrochrono, even though the device response [...]

I think WEC-Sim's new PTO/constraint extension feature is a much more robust way to implement initial displacements. How does this feature compare to HydroChrono? The extension feature is actually displacing the joint, whereas body.initial.displacement, pto.initial, etc are just moving the bodies in space and require an additional change to PTO pretension

PR #1369 (2024-11-13): Update forceInternalMechanics and powerInternalMechanics in PTO blocks
This PR resolves #1368. See discussion for details 

In the PTO library:
- redefine ``forceInternalMechanics`` to be ``F_total - F_actuation - F_constraint``
- redefine ``powerInternalMechanics`` to use the new ``forceInternalMechanics`` and the ``velocity`` signal

This PR does not yet remove powerInternalMechanics, though I think that is a worthwhile discussion per my comment in #1368