WEC-Sim/WEC-Sim #1372: [BUG] Two bodies have different x-coordinates in RM3 example

fanofdw, 2024-11-19:
**Describe the bug (REQUIRED)**
I checked the results of the RM3 example and noticed that the first column (x-coordinate) of output.bodies(1).position and output.bodies(2).position shows a significant difference. The maximum difference is 0.57 m. This seems unusual because the two bodies are constrained by a linear PTO (only relative motion in the z-direction is allowed).

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 10
 - [MATLAB/Simulink Release] 2021b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html):  v6.0, v6.1.1 （two versions have the same results）

**WEC-Sim file(s):**
[https://github.com/WEC-Sim/WEC-Sim/tree/main/examples/RM3](url)

**Expected behavior**
The x-coordinates in the first column of output.bodies(1).position and output.bodies(2).position should be equal or nearly equal.

dforbush2, 2024-11-19:
Hi @fanofdw 

From the seabed up, the constraints in the RM3 example are: 3DOF floating --> spar --> linear PTO --> float. The float and spar are thus constrained to be along the line-of-action of the linear PTO, but that PTO may not always align with the global Z direction because the 3DOF floating constraint will allow the spar, and everything connected to the spar, to pitch. If the spar pitches, the PTO will pitch, and so will the buoy. The X-position of the float and spar will thus be different, but if you check the pitch output.bodies(1).position(:,5), they should be identical owing to the translational constraint between them. 

This is different than the RM3_DD_PTO example in the applications repository because the constraint adjacent to the seabed is a Translational constraint. Thus, the spar cannot pitch, and so it will always have the same X-coordinate as the float.

fanofdw, 2024-11-20:
@dforbush2 Thank you so much for your rapid and detailed response. I have no more questions about this.

closed 2024-11-20 by fanofdw