WEC-Sim/WEC-Sim #1375: [BUG] rapid-accelerator mode recompiles each run when parallelized

rebeccamccabe, 2024-11-26:
**Describe the bug (REQUIRED)**
I am trying to use rapid accelerator mode with parallel computing toolbox. When I run wecSimPCT with no modifications, the sim is successful but it recompiles the accelerated target each time, which negates the point of the accelerator. I tried to follow [this](https://www.mathworks.com/help/simulink/ug/not-recommended-using-sim-function-within-parfor.html#brsk7nj) suggestion from mathworks to avoid the recompile. But I am getting an error with my modifications that I'm not sure how to resolve.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 10
 - [MATLAB/Simulink Release] R2022a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): I haven't pulled in a few months, I have commit d6f334f6ee0e70a598972566918f0b28f33128ca which is between 6.0 and 6.1. I confirmed that I do have the accelerator-related fix from 6.1 #1100 . 
 - Compiler: first I used the matlab default LCC. That gave a warning of unrecommended compiler before the error, so then I installed [minGW 6.3 from matlab](https://www.mathworks.com/matlabcentral/fileexchange/52848-matlab-support-for-mingw-w64-c-c-fortran-compiler). That took away the warning but not the error.

**WEC-Sim file(s):**
`wecSim.m` and `wecSimPCT.m`

**How to reproduce:**
Using `examples/RM3`, with the following wecSimInputFile, which is a minimum example with just 2 sims to be parallelized:
```
mcr.header = {'waves.height','waves.period','pto(1).damping'};
H = [1; 1];
T = [5; 8];
B = [1e6; 2e6];
mcr.cases = [H,T,B];
save('mcrTest.mat','mcr')

%% Simulation Data
simu = simulationClass();               % Initialize simulationClass
simu.mode = 'rapid-accelerator';        % Specify Simulation Mode ('normal','accelerator','rapid-accelerator')
simu.simMechanicsFile = 'RM3.slx';         % Simulink Model File
simu.startTime = 0;                     % Simulation Start Time [s]
simu.rampTime = 100;                   	% Wave Ramp Time [s]
simu.endTime=400;                       % Simulation End Time [s]
simu.dt = 0.1; 							% Simulation time-step [s]

%% Wave Information 
% Regular Waves  
waves = waveClass('regular');              % Initialize waveClass
waves.period = 999;                          % Wave Period [s]
waves.height = 999;                          % Wave Height [m]

%% Body Data
% Float
body(1) = bodyClass('hydroData/rm3.h5');          % Initialize bodyClass for Float      
body(1).geometryFile = 'geometry/float.stl';    % Geomtry File
body(1).mass = 'equilibrium';                   % Mass [kg]
body(1).inertia = [20907301 21306090.66 37085481.11];  % Moment of Inertia [kg*m^2]     

% Spar/Plate
body(2) = bodyClass('hydroData/rm3.h5');          % Initialize bodyClass for Spar/Plate
body(2).geometryFile = 'geometry/plate.stl';    % Geometry File   
body(2).mass = 'equilibrium';                   % Mass [kg]
body(2).inertia = [94419614.57 94407091.24 28542224.82];   % Moment of Inertia [kg*m^2]     

%% PTO and Constraint Parameters
% Floating (3DOF) Joint
constraint(1) = constraintClass('Constraint1'); % Initialize constraintClass for Constraint1
constraint(1).location = [0 0 0];                    % Constraint Location [m]

% Translational PTO
pto(1) = ptoClass('PTO1');                      % Initialize ptoClass for PTO1
pto(1).stiffness = 0;                                 % PTO Stiffness [N/m]
pto(1).damping = 999;                                 % PTO Damping [N/(m/s)]
pto(1).location = [0 0 0];                           % PTO Location [m]
```
Output when running wecSimPCT with no modifications - compilation happens twice:
[image]

Then to follow the mathworks suggestion, I made a modification to `wecSim.m`, adding a parameter to the `sim` line:
`sim(simu.simMechanicsFile, [], simset('SrcWorkspace','parent'), RapidAcceleratorUpToDateCheck='off');`
and in `wecSimPCT.m`, add the following line right before the `parfor`:
`rtp = Simulink.BlockDiagram.buildRapidAcceleratorTarget(load_system(simu.simMechanicsFile));`

Resulting error:
[image]

**Expected behavior**
I expect the accelerator compilation to be successful without error, and only run once instead of twice.

**Additional information:**
It seems like the error is related to [this variant control update](https://www.mathworks.com/matlabcentral/answers/1847923-variant-errors-when-upgrading-the-matlab-release), but I am not sure how to fix it, I am not familiar with wecsim's use of base vs mask workspace.

rebeccamccabe, 2024-12-03:
Thank you for the reply, I will aim to submit a PR later this week

kmruehl, 2025-02-05:
@rebeccamccabe are you still interested in submitting a PR?

kmruehl, 2025-03-10:
I'm going to go ahead and close this issue.

closed 2025-03-10 by kmruehl