WEC-Sim/WEC-Sim #1381: [BUG] External Force and Torque block (mask) do not have a parameter named 'SimscapeInstrumentationLogging.'

Juan-Guerrero-Fernandez, 2024-12-13:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
There are some errors with the SimscapeInstrumentationLogging parameter.

When TestWECCCOMP/testWECCCOMP_MPC are run, a number of errors appeared:

-External Force and Torque block (mask) does not have a parameter named 'SimscapeInstrumentationLogging' 
-External Force and Torque block (mask) parameter 'SimscapeInstrumentationLogging' is read-only

-Transform Sensor block (mask) does not have a parameter named 'SimscapeInstrumentationLogging'
-Transform Sensor block (mask) parameter 'SimscapeInstrumentationLogging' is read-only

And the same for several more blocks

**Workstation information (REQUIRED):**
 Operating System: Windows 10 Enterprise
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html):  2023b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
Run TestWECCCOMP.m, from WEC-Sim_Applications\WECCCOMP

**Expected behavior**
Simulation should run without problems. A month ago, it was working, here is the result of running [WECCCOMP_Nonlinear_Model_Predictive](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/WECCCOMP/WECCCOMP_Nonlinear_Model_Predictive)

[image]
 

**Screenshots:**
Current error:

[image]

**Additional information:**
I believe it is related with the issue #1368
ptoInternalMechanics incorrect when a PTO has non-zero pretension or equilibrium position

dforbush2, 2024-12-17:
This is a pervasive issue and also occurs on other WEC-Sim examples not just applications cases. At this time it appears that Simscape Explorer data logging has been inadvertently enabled on some traces and/or on a library-wide setting. Speculation: this may be a version issue and in conversion to 2020b libraries. In the mean time please revert to an earlier release of WEC-Sim: this error is my top priority and either it or I will no longer exist by break of day.

dforbush2, 2024-12-17:
@Juan-Guerrero-Fernandez as a stop gap:
1) place a debug point on line 43 of wecSim.m 
2) >>wecSim
3) open your *slx model.
4) run your model using the green arrow in the simulink GUI.

For some reason, this works and avoids the error in SOME tested models. Hopefully this will get you rolling while we figure this out. 

This implies that this error occurs during run-from-matlab model initialization.

dforbush2, 2024-12-17:
***THIS ISSUE DOES NOT APPEAR ON WEC-SIM/DEV***
@Juan-Guerrero-Fernandez another workaround for you and others in the mean time is to switch to the developer branch of WEC-Sim

I'm not sure why this is, but the binary file modifications in seem to be the genesis of this issue #1373
@MShabara and @akeeste any insight would be super helpful since its really hard to see the exact change in the binary file. Perhaps you remember and can explore a different approach? 

This issue seems to arise in all cases where a model relies on a block with an _intentionally_ broken library link i.e. the model this user is testing, several other of the WS Applications runs. For other developers attempting to replicate with a very simple case:
1) open examples/RM3
2) disable any library link
3) run wecSim
 
In this case the above work-around (debug point in wecSim, run model from the simulink gui) still works.

Weirdly, the error messages appear specifically for blocks whose library links remain intact. 

Because #1373 was itself a bug fix, I will not revert it without first consulting @MShabara and @akeeste

jtgrasb, 2025-01-16:
Closing this as it is resolved in the latest WEC-Sim release.

PR #1388 (2024-12-23): Resave body library to resolve simscape logging error
I did not change anything... I just resaved the body library and it logged as a change to the library and resolved the simscape logging error that was occurring with some of the WEC-Sim Applications cases. This should resolve #1381 and #1386.

@MShabara This should be merged before pushing dev to main.
closed 2025-01-16 by jtgrasb