WEC-Sim/WEC-Sim #1385: [BUG]Question about the calculation non-linear excitation loads

TianyuanWangi, 2024-12-14:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
Hi,

I recently checked out the nonlinear calculation module for floating bodies. I found the wave spread is not considered in the nonlinear module, both in waveElevation.m and nonFKForce.m.

Is nonlinear calculation not able to consider the wave spread? Or special treatment elsewhere?

Kind regards,

Tianyuan

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): [e.g., 2022b, 2023a, 2023b, 2024a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): [e.g., v5.1, v6.0, v6.1]

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
_How can we reproduce the error? Describe the steps to reproduce the error._

**Expected behavior**
_What is the excepeted behavior? Describe the expected behavior._

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screenshot of the error message from the MATLAB command window._

**Additional information:**
_Provide any other additional information._

dforbush2, 2025-01-11:
Hi @TianyuanWangi 

You are correct. The non-linear excitation forces do not include wave spread and thus should not be used (at this time) for multi-directional waves expressed using the wave spread function. 

I will address this with a pull request and leave this issue open until then.

kmruehl, 2025-04-03:
@dforbush2 can we close this issue now that #1439 is merged?

closed 2025-04-03 by dforbush2