WEC-Sim/WEC-Sim #1394: Improve variable hydro radiation force calculation

akeeste, 2025-01-13:
Per some quick comments today from Giorgio and @ryancoe, I think we need to revisit how the convolution integral method of the radiation force is calculated:

Currently what we do (with variable hydro and fixed time-step to simplify writing the math) is:

$$F_{rad}(t_i) = \int^{t_i}_0 v(\tau) \cdot K_r(t-\tau) \cdot d\tau$$

$$F_{rad}(t_i) \approx dt \cdot \left( v(t_i) \cdot K_r(0) + v(t_{i-1}) \cdot K_r(dt) + v(t_{i-2}) \cdot K_r(2*dt) + ... \right)$$ 

But, shouldn't the time history of the IRF coefficient ($$K_r$$), reflect that state of the device that made that contribution at that time? So if we switched from state B to state A at $$t_{i-1}$$, the radiation force should be:

$$F_{rad}(t_i) \approx dt \cdot \left( v(t_i) \cdot K_{r,A}(0) + v(t_{i-1}) \cdot K_{r,A}(dt) + v(t_{i-2}) \cdot K_{r,B}(2*dt) + ... \right)$$

akeeste, 2025-01-13:
The necessity of this change is dependent on how much the varying state changes over ``simu.cicTime``. Even with fine discretization, the state could change significantly over the convolution time and necessitate updated coefficients.

If the last equation above is the most accurate way to approach this issue, it shouldn't be too hard to implement. In the convolution integral calculation, we need to record the IRF coefficient history like we do the velocity history

dforbush2, 2025-01-13:
pulling from our other dialog, the issue doesn't "go away" if we have a fine discretization between hydrodynamic data sets because the data buffer for the convolution integral is (or should be) sufficiently long that we may see a significant change in the device geometry over this time. 

Practically though, how do we do this? As a straw man: I propose an additional structure probably below the (hf) substructure of the body that contains an IRF "surface" that we calculate during the pre-processing stage, basically we calculate each IRF function over CIC-time, where the additional axis is the state variable that we presently use to switch between the (hf) substructures. We then revise the IRF calculation to pull the Kr coefficients from a (probably tabular gridded interpolant) representation of this table.

akeeste, 2025-01-13:
Yeah I was thinking further over lunch, the implementation isn't quite as easy as tracking the velocity history. We need the entire set of Kr coefficients over all states at each time step. 

@dforbush2 I think you're right on. And agreed about redundancy--that Kr surface could get large and there's no need to duplicate it as an hf substructure.

dforbush2, 2025-01-13:
in the block diagram, we also will want to loop back a Kr buffer updating every timestep (or at a separate, less frequent rate we specify). This will allow us to perform a single 1x1 table lookup, interpolation, or calculation operation each time step.

akeeste, 2025-02-05:
This same logic could also apply to the excitation convolution if we use it. Check on when this could be the case

akeeste, 2025-02-05:
The excitation convolution is only used in the case of a user defined excitation force.

kmruehl, 2025-04-03:
@akeeste can we close this issue now that #1401 is merged? If not, what else needs to be done to close this issue? Thanks

akeeste, 2025-04-03:
@kmruehl #1401 is unrelated. #1411 is the related PR. I believe @jtgrasb wanted a second review on it as there were a lot of library changes

akeeste, 2025-04-30:
Closed by #1411

closed 2025-04-30 by akeeste