WEC-Sim/WEC-Sim #1398: [Feature]

Gusmano-2-OSU, 2025-01-16:
**Describe the requested feature (REQUIRED):*
A flag in in the `simu` class that keeps the parallel pool running after a successful run of `wecSimPCT`.

**Related issue:**
My WEC-Sim workflow often involves simulating a batch of cases via `wecSimPCT`, then assembling the results in a convenient format by the parallel computing toolbox. Currently, I run the simulations, shut down the parallel pool, reinitialize the parallel pool, then assemble the results, wasting time as MATLAB closes and reopens the pool.

**Describe alternatives solutions:**
Of course, we could simply wait.

**Related  application:**
N/A

**Additional information:**
N/A

dforbush2, 2025-01-25:
Try commenting this line of wecSimPCT:

% delete(gcp);

That should keep the parallel pool active. But! It will still timeout on its own if left idle for too long.

Gusmano-2-OSU, 2025-01-29:
Hey @dforbush2, thanks for the reply! Sorry for my late response 😅. Technically speaking, this _does_ resolve the issue I mentioned, but I don't think I articulated myself well in presenting the issue. Suppose you're interested in running another batch of cases off the pre-existing parallel pool from our last run of `wecSimPCT`. When I do this, I get the following error:

```
Error using [parallel.Cluster/parpool](matlab:matlab.lang.internal.introspective.errorDocCallback('parallel.Cluster/parpool', 'C:\Program Files\MATLAB\R2024b\toolbox\parallel\cluster\+parallel\@Cluster\parpool.m', 79)) ([line 79](matlab: opentoline('C:\Program Files\MATLAB\R2024b\toolbox\parallel\cluster\+parallel\@Cluster\parpool.m',79,0)))
Unable to create parallel pool because creating concurrent parallel pools in the same MATLAB session is not supported. To delete the existing pool, use
[delete(gcp('nocreate'))](matlab:delete(gcp('nocreate'))).

Error in [wecSimPCT](matlab:matlab.lang.internal.introspective.errorDocCallback('wecSimPCT', 'C:\Users\paulg.000\Source Code\Vocational\Ohio State\WEC-Sim\source\wecSimPCT.m', 33)) ([line 33](matlab: opentoline('C:\Users\paulg.000\Source Code\Vocational\Ohio State\WEC-Sim\source\wecSimPCT.m',33,0)))
parpool(p); % open the pool
^^^^^^^^^^
```

I think the point I'm trying to get at is this—opening and closing the parallel pool takes time. Ideally, I'd be able to open a single parallel pool when I start working with WEC-Sim in a larger workflow, and close it when I'm done. Implementing this would be a quality of life feature that could help save time in between different steps for users that use `wecSimPCT` as part of a larger workflow involving the parallel toolbox, like I do. Of course, I could imagine situation where users _would_ want the parallel pool closed immediately after `wecSimPCT`'s completion, so I suggested implementing it as a flag.

**TL;DR** It would be convenient to use pre-existing parallel pools with `wecSimPCT` to save time that otherwise would have been used opening and closing a new parallel pool.

Does that make sense? Feel free to ask more questions if I could've articulated myself better!

dforbush2, 2025-02-05:
Hi @Gusmano-2-OSU 

I like the suggestion for a flag. I definitely encourage you to take it on and make a PR with this modification! I'll also do it, but probably won't get to this quickly. 
With parpool, MATLAB creates the pool based on your system specification, so this should be pretty robust. (Irrelevant aside: I don't think this is true with parsim)
We can 
1) add a check for an existing pool before creation, and use that for the simulations if it does. A lazy way is a try-catch. A better way is something like 
`isempty(gcp('nocreate'))`
2) add a flag to simulationClass that gives a user the option to either delete the parallel pool at the end of wecSimPCT or not.

kmruehl, 2025-02-19:
@Gusmano-2-OSU Thank you for submitting a PR. @dforbush2 will review it and close this issue once the PR has been merged.

kmruehl, 2025-03-11:
Closing now that https://github.com/WEC-Sim/WEC-Sim/pull/1418 is merged

closed 2025-03-11 by kmruehl