WEC-Sim/WEC-Sim #1400: [BUG] Applications breaking.

akeeste, 2025-01-16:
Some application cases are breaking in v6.1.2. Given our current release is buggy, this should be a high priority to resolve. I just reran all Applications tests using both main branches and some are still failing. Results [here](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600)

[MOST](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731020055)- these regression tests have also failed previous. It's unclear why an identical case results in some slightly different outputs. We can reset the regression data or try to test against a case without rerunning any processes with optimization or randomness. We've had prior discussions on this point, I'll try to find and link to them.

[Mooring/MoorDyn](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731021137) - some drastic crash, crashing MATLAB entirely. It's hard to tell what's going on from the log, but it has failed on a 2nd retry now.

[Morison element](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731022001#step:9:37) - @dforbush2 @jtgrasb The full directional wave spectra #1332 seemed to break other tests running with wave spreading due to the change [here](https://github.com/WEC-Sim/WEC-Sim/blame/6a143ce063b6f4613831f5664018d62a6ec44dd2/source/objects/waveClass.m#L819). Any idea what's happening? Is there a test case or application for a full directional wave spectra case?

[Multiple wave headings](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731023609) - failing while processing a bug signal. I'm guessing its related to variable hydro buses. @akeeste will investigate

[Passive Yaw](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731027609#step:9:271) - the regression test is failing due to differences in results. My guess is that this related to the fixes for wave spreading in #1289 (also see issues #1367 #1377). Not sure how to confirm this. If this is the case and the current simulations are more accurate, we need to update the regression data being tested against on the Applications repo

akeeste, 2025-01-16:
Source fixes in #1401
Application fixes in WEC-Sim/WEC-Sim_Applications#80

akeeste, 2025-01-16:
On passive yaw--I thought the difference in yaw displacement and yaw torque would be coming from the fix to the spreading function. The passive yaw case does use spreading, but it only uses one wave direction with a single spread value of `waves.spread=1`, so that particular bug should not be affecting that case.

akeeste, 2025-01-22:
The Paraview RM3 test should be failing because it uses MoorDyn too. I've updated the logic in the applications PR so that this case runs correctly. If it fails, it can be fixed with the same solution as the MoorDyn test (TBD). if it passes I'll use that to fix the MoorDyn test

jtgrasb, 2025-01-22:
@RyanDavies19 We are having some issues with the MoorDyn dll for some reason. As Adam pointed out, it seems to [crash MATLAB entirely](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12776912600/job/35731021137). I verified that this works locally and I also found that it works fine with the previous commit of WEC-Sim/MoorDyn (meaning it seems to be related to the recent update to 2.3.7) - see [PR on my fork](https://github.com/jtgrasb/WEC-Sim_Applications/actions/runs/12913390630/job/36010618386?pr=1) where MoorDyn test using 2nd to last commit doesn't crash MATLAB. I am having a hard time debugging this since it doesn't crash locally, but wanted to check if you have any thoughts?

Also, @akeeste the mooring matrix test seems to be failing too due to the 'SimscapeInstrumentationLogging' error.

jtgrasb, 2025-01-22:
Based on my testing, it does not have to do with any new commits made on the WEC-Sim Applications repository. It seems to be the new DLL that is causing the error (from [last PR](https://github.com/WEC-Sim/MoorDyn/commit/9a7e43959098334c2806f83635eb4349d94903e5)) because it does not crash when testing the old DLL. 

We don't have a Linux test for MoorDyn because we were still working on the .so file, and we don't test WEC-Sim with Mac by default but could try to test this locally.

RyanDavies19, 2025-01-22:
@jtgrasb, can we try this with MoorDyn 2.3.8, the most recent release? The DLL in there now is 2.3.7, and if you remember we had issues with the API interface we were using. We commented out the troublesome code: https://github.com/WEC-Sim/MoorDyn/commit/9a7e43959098334c2806f83635eb4349d94903e5#diff-8d5fcb6be7e83fa58a6d7139ecea1039fbca895935bc81d138f62d19947cdbb3L147

But v2.8.3 has a more permanent fix to it: https://github.com/FloatingArrayDesign/MoorDyn/pull/280.

akeeste, 2025-01-22:
The discrepancy in the passive yaw regression test:
- last passed [here](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12279299732/job/34263218376), using WEC-Sim dev commit 1c62b6d
- first failed [here](https://github.com/WEC-Sim/WEC-Sim_Applications/actions/runs/12279629534/job/34264709305), using WEC-Sim dev commit a8b9dc0

@dforbush2 Based on the dev log, it looks like the discrepancy is coming from #1332. Seemingly related to how the wave direction is interpolated across +-180 degrees. Were there any changes to pre-processing in #1332 that would affect nonlinear yaw, even if the new wave type is not used?

jtgrasb, 2025-01-22:
@RyanDavies19 Just tested it with 2.3.8 and looks like its still causing the same issue where it [crashes MATLAB](https://github.com/jtgrasb/WEC-Sim_Applications/actions/runs/12918433919/job/36026985102?pr=1). I am going to test a few versions out to see which version of MoorDyn the issue shows up in, and for now go back to a dll version before the issue arises.

akeeste, 2025-01-30:
@jniffene @MShabara @kmruehl I was digging through the MOST regression tests again today. I don't have a solution but am recording my findings for us to discuss:
- I have tested with several versions of MATLAB (R2023a, R2023b, R2024a) and have not found any differences between versions
- I noticed that the differences found by the regression test changes significantly when the ``mostIO`` pre-processing functions is re-run first. When running the application using our current source but not rerunning the MOST pre-processing, the tests fail but are largely explainable.

1. In the case of using the committed pre-processing files in `mostData`, the discrepancies are fairly isolated. For many of the points in time for which the new solution is not equivalent to the previous, they are:
   -  turbulent_bodyPitch errors but is very close to our arbitrarily defined tolerance. 30 points have an error of 0.0101%, over the tolerance of 0.01%.
   - turbulent_bladePitch errors at a single point with an error of 0.01005%, over the tolerance of 0.01%
   - turbulent_towerBaseLoad errors at 2, non-consecutive points with an error of 16.6% and 4%, over the tolerance of 3%. BUT I think it is important to recognize that the magnitude of the tower base load at those points is actually relatively small (~1e4) compared to the typical magnitude of that load during the simulation: ~1e8. I also find it very strange that a single point has 16% error but no other point near it in time has over 3% error.
   - turbulent_towerTopLoad The same scenario and logic as in the case of turbulent_towerBaseLoad applies here

2. In the case where I rerun the MOST pre-processing, the discrepancies are larger and more widespread across each variable, though certainly from the perspective of visualizing the tests the variables look to follow identical trends.

@jniffene have you been able to narrow the source of the test discrepancies?

I'm personally quite stuck on how to proceed further. I'm leaning towards accepting the current regression tests have a very tight tolerance and fail. We could then start fresh and overhaul the MOST regression tests entirely when we pull in the new version of MOST, which will be more accurate than the current implementation.

closed 2025-03-11 by kmruehl

akeeste, 2025-03-12:
Most of the applications should now be passing. The outstanding failures work locally (windows) or with another MATLAB version. 4 application tests are currently skipped:
- Mooring/MoorDyn, Paraview RM3 MoorDyn - MoorDyn fails very sporadically on the runner. It's unpredictable but works locally
- Paraview OSWEC - fails on the runner, works locally
- Mooring/MooringMatrix - rapid-accelerator has specific issues on R2024b. This seems like a MATLAB-based problem and we should reach out to them

PR #1401 (2025-03-01): Bug fixes for applications
PR to address the bugs appearing in our applications cases, as described in #1400. Closes #1400