WEC-Sim/WEC-Sim #1405: [Theory or Implementation] Question about how pressures are calculated

brittlyd, 2025-01-28:
**The development team requests that before submitting a question on the theory and implementation behind WEC-Sim to first please:**
1. Read through our [WEC-Sim Documentation](http://wec-sim.github.io/WEC-Sim/man/theory.html) with additional description found in our [Advanced Features Section](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html)
1. Review MATLAB source code ([functions](https://github.com/WEC-Sim/WEC-Sim/tree/main/source/functions), [objects](https://github.com/WEC-Sim/WEC-Sim/tree/main/source/objects)) and the Simulink blocks ([libraries](https://github.com/WEC-Sim/WEC-Sim/tree/main/source/lib))
1. Review the curated [WEC-Sim Examples](https://github.com/WEC-Sim/WEC-Sim/tree/main/examples) and the [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications)
1. Review the [WEC-Sim issues page](https://github.com/WEC-Sim/WEC-Sim/issues) which allows you to search past resolved issues or questions that might help answer your own question.

**Note: italicized text below is include as an example and should be updated before submission. If you feel any section is not applicable to your request, please replace with 'N/A' rather than delete a section.**

**Is your question request related to a problem? Please describe.**
This is a general question that applies to all problems.  

**Describe the theory or implementation approach that you have a question**
How is the pressure calculated in output.bodies.cellPressures_waveLinear and output.bodies.cellPressures_waveNonLinear? Is it calculated from potentials solved using BEM? 

**Personal project context**
I am trying to compare the pressure measurements in the OSWEC example to those calculated using a different theory (from [this paper](https://www.sciencedirect.com/science/article/abs/pii/S0997754613000174)) and want to make sure I am comparing the correct values. 

**Describe the type of conclusion or resolution you are looking for**
I am hoping to understand how pressure is calculated for the output.bodies.cellPressures_waveLinear and _waveNonLinear vectors.

**Additional context**
_Add any other context or screenshots about the feature request here._

jniffene, 2025-01-30:
Hi thank you for your question! WEC-Sim determines pressure distributions only for nonlinear bodies, so for now "output.bodies.cellPressures_waveLinear" results in a matrix of zeros. But the nonzero values of "output.bodies.cellPressures_waveNonLinear" are determined via the functions in the nonFKForce.m file.

kmruehl, 2025-02-19:
@brittlyd I'm going to go ahead and close this issue. Please reopen it if you have any further questions.

closed 2025-02-19 by kmruehl