WEC-Sim/WEC-Sim #1406: [BUG] Main branch failing for R2022a and R2022b on Ubuntu

jtgrasb, 2025-01-29:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
The [main branch of WEC-Sim is failing](https://github.com/WEC-Sim/WEC-Sim/actions/runs/12776910751/job/36370508773) on Ubuntu for R2022a and R2022b. I narrowed down the error to be when `set_param(0, 'ErrorIfLoadNewModel', 'off')` is run. For some reason, this command is not executing correctly for this OS and MATLAB version.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Github Actions; ubuntu-latest, R2022a and R2022b

**WEC-Sim file(s):**
`run-tests-main.yml`

**How to reproduce:**
[_How can we reproduce the error? Describe the steps to reproduce the error._](https://github.com/WEC-Sim/WEC-Sim/actions/runs/12776910751/job/36370508773)

**Expected behavior**
WEC-Sim tests [ran without error](https://github.com/WEC-Sim/WEC-Sim/actions/runs/12776163725) as of 2 weeks ago.

**Screenshots:**
[image]

**Additional information:**
N/A

kmruehl, 2025-02-05:
@kmruehl update tests for main and dev to be the latest 4 or 5 MATLAB releases

kmruehl, 2025-03-01:
resolved by https://github.com/WEC-Sim/WEC-Sim/pull/1420

closed 2025-03-01 by kmruehl