WEC-Sim/WEC-Sim #1408: [BUG]: Interpolation error when using full directional wave spectra

laucrom, 2025-01-30:
**Describe the bug**
When running WEC-Sim with an (imported) fully directional wave spectrum (spreading is frequency dependent) with BEM data ranging from 0° to 360°, an interpolation error appears stating that my data is not in ascending order, although it is. I took a look at the code and I think that the use of the 'wrapto180' might be a problem as e.g. 270° will become -90° resulting in a non-ascending data order.

**Workstation information:**
Operating System: Windows 10
[MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2023b
[WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1.2

**WEC-Sim file(s):**
The error is found in 'bodyClass.m'. Specifically, in the interpolation routine where BEM data is interpolated to all wave directions to determine excitation forces.

**How to reproduce:**
I made an arbitrary test spectrum, as in issue #1407. For some reason, it seems I cannot upload my test case here. The only difference is that in this case my BEM data ranges from 0° to 360° instead of -180° to 180° in issue #1407.

**Expected behavior**
The expected behavior is that WEC-Sim completes the simulation without error.

**Screenshots:**
The error message is the following:

[image]

laucrom, 2025-01-31:
I am using Capytaine. I redid the BEM simulations and made another test case, which now returns a different error and several warnings that my BEM data does not span 360°, although it does. It also returns a warning stating something about passive yaw, but I did not activate passive yaw. You can find the test case here: 

[NewTest_BEMrange_FullDirectional.zip](https://github.com/user-attachments/files/18616037/NewTest_BEMrange_FullDirectional.zip)

This is the full error message/warnings:

[image]

[image]

jtgrasb, 2025-02-03:
@laucrom Thanks for sharing the updated BEM data. The error you are getting now is because your BEM data has both a datapoint at 0 degrees and at 360 degrees. These are effectively the same and the wrapTo180() function wraps both to 0 and causes the error. I used this code to fix and verify that it works by removing the 360 deg datapoint.

```
hydro = readH5ToStruct("1_WEC_Data_d1p10_IRR_AllDirections_0360_d15.h5");

hydro.theta = hydro.theta(1:24);
vars = ["ex_ma","ex_ph","ex_re","ex_im","sc_ma","sc_ph","sc_re","sc_im","fk_ma","fk_ph","fk_re","fk_im","ex_K"];

for ii = 1:length(vars)
    hydro.(vars(ii)) = hydro.(vars(ii))(:,1:24,:);
end

hydro.file = 'cutOut360';
writeBEMIOH5(hydro);
```

This should resolve your errors for now, but I think on our side, we should also address these in the code by:
- If the angles are not in ascending order, convert them to ascending order (and show warning)
- If an angle is repeated, remove the duplicate (and show warning).
@dforbush2 Interested to hear your thoughts on the above proposed changes.

As for the warnings you are getting:
- We need to fix the if statement for the passive yaw options by putting parentheses around the "or" options.
- Our method of determining the difference between each direction is a bit strange and produces a warning even though your directions are well spaced. We may need to adjust this as well.
- The last warning also needs to be adjusted as it is only giving the warning after the wrapTo180() function.

dforbush2, 2025-02-11:
hi @laucrom 
be advised there may be an outstanding bug in full directional wave spectra that may be giving erroneous excitation calculations. working on it now.

dforbush2, 2025-02-11:
* adding parentheses grouping the "or" statements on line 953 will fix the passive yaw warning related to BEM not spanning a full rotation.
* agree this is strange and needs revisions...it's difficult to come up with a good flag for all possible "bad" BEMs but this one is not working as intended.
* I think the problem here is that <360 is too strict a criteria: without repeated points (which you don't want as per error above), your wrapTo180 call will output something a vector with endpoints something like [-180 + x, 180 -y], where hopefully x xor y is zero, implying you'll always trigger this warning. Recommend reducing to, say, <350 or so.

dforbush2, 2025-02-12:
@laucrom see #1288 

bug found. The spread function should not be squared in irregExcFullDir. (or irregExcF, for that matter)

will push bug fix ASAP