WEC-Sim/WEC-Sim #1413: [BUG] Full directional spectrum excitation does not seem to be accurate

jtgrasb, 2025-02-10:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
When using the full directional spectrum implementation, the excitation is lower than expected. I believe the issue is because the spread values are being squared when they should not be.

[image]

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2023b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1.2

**WEC-Sim file(s):**
WEC-Sim\source\functions\simulink\model\irregExcFullDirF.m

**How to reproduce:**
N/A

**Expected behavior**
The spread value should not be squared. See below (from @dforbush2 and I's paper) where $P(\omega_n, \Theta_m)$ is the portion of spectral energy in each directional bin for each frequency and is within the square root and not squared. This is also consistent with [this paper](https://www.researchgate.net/profile/Emilio-Faraggiana/publication/335834302_Multidirectional_waves_and_time_domain_perturbed_field_visualization_of_the_WaveSub_device/links/5d7f3f814585155f1e4f5ec4/Multidirectional-waves-and-time-domain-perturbed-field-visualization-of-the-WaveSub-device.pdf).

[image]

**Screenshots:**
N/A

**Additional information:**
@dforbush2 I'm struggling a bit to identify how this is different from Issue #1288, but it seems that for some reason the irregular wave spread weights should be squared while the full directional spread weights should not be squared.

dforbush2, 2025-02-11:
Thanks for looking at this @jtgrasb 

I've reviewed the implementations and I think that the distinction is between the amplitude spectrum $A(\omega)$ and the wave spectrum density $S(\omega)$, commonly and confusingly just called the wave spectrum (at least according to Wanan and Sheng, 2018, equation 5-6).

where they are related by: 

$S(\omega_n) = \frac{abs(A(\omega_n)^2}{2 \Delta \omega}$

In our paper, the initial wave data was given as $S(\omega)$  with units of m^2/Hz (we convert $\Delta \omega$ to Hz too)

That the spread vector summed over all directions (at each frequency, or generally for other wave types) must sum to one is defined in terms of spectral density, or,

$S(\omega,\beta) = S(\omega) G(\omega,\beta)$

where

$ \int_{-\pi}^{\pi} G(\omega, \beta) =1.

On inspection of our waveClass, calculating elevation for, say, irregular waves:
```
 for iw = 1:length(obj.omega)
                            Z = Z + sqrt(obj.amplitude(iw)*obj.spread(idir).*obj.dOmega(iw)) * cos(-1*obj.wavenumber(iw)*Xt + obj.omega(iw)*t  + obj.phase(iw,idir));
end
```

You can see that what has been called "waves.amplitude" is, aside from a factor of 2, a wave spectrum density, $S\omega$. I now believe that the correction made to #1288 may be in error. But I'm now very confused about that factor of 2.

dforbush2, 2025-02-12:
@jtgrasb you're correct and the bug was introduced (by me :( ) in #1288. 

it should NOT be squared. This motivates refactor of our waveClass and simulink functions because they are very over-complicated.

Try removing that square, and you should see near-equivalence between this and the test case.

jtgrasb, 2025-02-12:
@dforbush2 I spent this morning looking into this and running some tests. 

1. I cannot find the exact reason we added #1288. I remember at the time that the case with spread included seemed to have more energy in the system than the case without spread. I now checked and verified that removing the square in the excitation case does not seem to lead to any lack of conservation of energy. Perhaps there was something wrong with our approach to using spread at the time that caused the discrepancy. It would probably be valuable to have a test case to document the conservation of energy with spread included though.
2. After also getting rid of the square in irregExcFullDirF.m, I tested it out. For the full directional spectrum, I ran one case setting the waves.phaseSeed = 1, then used those phase values to build the pre-calculated excitation force. The results are similar for the heave direction, but the surge direction response is opposite. I checked for a sign difference on the mean drift forces but couldn't find the culprit. 
[image] [image]

I've also included the test cases below. The obsData/extractDataWave.m script is used to create both the full directional spectrum and the excitation timeseries.

[pioneerDirSpreadTest.zip](https://github.com/user-attachments/files/18771407/pioneerDirSpreadTest.zip)

dforbush2, 2025-02-12:
with the exception of the mean drift discrepancy, fixed by #1417

dforbush2, 2025-02-13:
thanks to @jtgrasb  who spent the day looking at this with me:

The mean drift (and possibly other) discrepancies are following from how we wrap BEM results. In WEC-Sim, we typically use sort(wrapTo180(yourBEMDirections)) to build tables for interpolating as-needed between BEM-provided directions and the direction you may see in your simulation (a combination of yaw angle and wave direction). This is NOT a robust approach because it does not handle BEM populated ~ [-180 180] and BEM populated ~ [0 360] equivalently using spline interpolation. 

A basic example (this is far too coarse a discretization for a usable BEM file, just for sake of explanation)
 sort(wrapTo180([0:90:270])) =  [-90: 90: 180];

When we use interp2(..., 'spline'), points outside the included range will be extrapolated via spline from the nearest included points. In our example, this means, we're "extrapolating" in 2 physically different regions: the [0:90:270] data between 270 and 360 (the lower right quadrant) and the wrapped data between 180 and -90 (the bottom-left quadrant)

As you can tell, this is a bigger problem for coarser BEM, but is an easily improvable system, we just need to wrap the opposite endpoint in our constructed tables so that extrapolation is never performed.

jtgrasb, 2025-03-03:
This issue was resolved by #1417 (spread squared fix) and #1419 (interpolation/wrapping fix). Closing now.

closed 2025-03-03 by jtgrasb