WEC-Sim/WEC-Sim #1422: [Support]PTOSIM parameter definition

jccjay, 2025-02-20:
**Please review the following prior to submitting an issue:**
1. Read through the [WEC-Sim Documentation](https://wec-sim.github.io/WEC-Sim), especially information in the [Advanced Features](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html) section.
2. Review WEC-Sim ([source code](https://github.com/WEC-Sim/WEC-Sim/tree/main/source)
3. Review [WEC-Sim examples](https://github.com/WEC-Sim/WEC-Sim/tree/main/examples) and [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications).
4. Review prior [WEC-Sim issues](https://github.com/WEC-Sim/WEC-Sim/issues), search past issues and discussions that may help answer your question.

**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the question (REQUIRED):*
I studied the official case of ptosim, but I still have questions about the definition of the following parameters.

> 

```
ptoSim(1) = ptoSimClass('adjustableRod');
ptoSim(1).adjustableRod.crank = 3;
ptoSim(1).adjustableRod.offset = 1.3;
ptoSim(1).adjustableRod.rodInit = 5;
```

```
ptoSim(2).hydPistonCompressible.xi_piston = 35;
ptoSim(2).hydPistonCompressible.pistonStroke = 70;
```
I noticed that #602  had asked a similar question, but the link in the text was invalid.

And in this case, whether the piston stroke should be less than the rodInit.

[image]

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2023b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.0

**WEC-Sim file(s):**
N/A

**How to reproduce:**
N/A

**Expected behavior**
N/A

**Screenshots:**
N/A

**Additional information:**
N/A

jleonqu, 2025-03-11:
Hello @jccjay 
If I am correct, you are looking at the ["OSWEC_Hydraulic_PTO"](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/main/PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO) example on the Application repo. In that example, we are using the “Rotary to Linear Adjustable Rod” block. This block can be use when the cylinder rod of the hydraulic PTO is connected to the adjustable rod. like in the figure below:

[image]

The variable rodInit is the initial length of the rod, and the variable pistonStroke is the maximum length of the rod when it's fully extended.

I hope this can be useful for you!

Best,

Jorge Leon

kmruehl, 2025-04-03:
@jccjay does this resolve your issue?

closed 2025-04-08 by jccjay