WEC-Sim/WEC-Sim #1424: [Theory or Implementation] Using elevationToSpectrum.m

Gusmano-2-OSU, 2025-02-21:
**Is your question request related to a problem? Please describe.**
I'm trying to develop a custom wave to test the robustness of an algorithm. I've done this in the time domain as an elevation time series, but the documentation suggests converting to a time-equivalent spectrum for better results, and provides `elevationToSpectrum.m` to help this.

**Describe the theory or implementation approach that you have a question**
I'm trying to use `elevationToSpectrum.m` to convert my time-domain elevation import to a spectrum import. It outputs `frequency`, `amplitudeSpectrum`, `phaseSpectrum` and `complexSpectrum`. The `spectrumImport.mat` file needs frequency data, spectral energy density, and (optionally) phase information.

The frequency and phase information seem easy enough to get immediately from the function, but I think I'm missing some theoretical background to get spectral energy density. From my investigating, it seems like it's proportional to the square of the amplitude spectrum, but I've seen different definitions. I'm asking this question hoping for some clarification. I've looked into the example provided in WEC-Sim, and it's not clear to me how the user got the `spectrumData.mat` file from the `elevationData.mat` file.

**Personal project context**
Testing the robustness of an algorithm using a custom wave.

**Describe the type of conclusion or resolution you are looking for**
I'm unclear on how to take the information output from the `elevationToSpectrum.m` function and put it into the format needed for a spectrumImport, especially with regards to spectral energy density. Suggestions would be really appreciated!

**Additional context**
N/A

dforbush2, 2025-02-28:
@Gusmano-2-OSU your confusion is understandable because we're a bit inconsistent with our terminology and our usage is not always well aligned with textbook definition.

According at least to Sheng and Aggidis, "wave spectrum" implies a wave spectral density with units of meters^2/Hz. The direct output of a Fourier transform of time-domain wave elevation history, that textbooks call an Amplitude Spectrum will have units of meters. 

If you check out line 53 of elevationToSpectrum you'll see:

`amplitudeSpectrum = (abs(P1).^2)./(2*df);`

So the output amplitude spectrum has units m^2/Hz. So really, I should rename "amplitudeSpectrum" to "waveSpectrum". This aligns with the following canonical conversions

$$S(\omega_n)=\frac{|A(\omega_n)|^2}{2\Delta \omega}$$

Where $A(\omega_n)$ is the amplitude spectrum (m) and $S(\omega_n)$ is the wave spectrum (m^2/Hz)

***NOTE*** in the general case, waves.spectrum as actually NEITHER an amplitude spectrum or a wave spectral density. Its just
2 * amplitude spectrum. Which is very confusing.

We are talking a lot internally about refactoring waveClass and all supporting functions to use more intuitive naming.

Gusmano-2-OSU, 2025-02-28:
Hey @dforbush2, thanks a lot for this clarification! I'll mark this issue as closed and reopen it if I have any more questions!

closed 2025-02-28 by Gusmano-2-OSU

dforbush2, 2025-02-28:
Hope the above answers your questions. The short answer is if you input data to elevationToSpectrum is an elevation time history in meters, and your time vector is in seconds, then the outputs of this function will be appropriate to directly build a file for spectrum import.

Gusmano-2-OSU, 2025-02-28:
Ope, I can't reopen it 😅. I guess I'll just reach out if I have more questions!