WEC-Sim/WEC-Sim #1427: Installation problems in MATLAB

EnochBadu, 2025-03-05:
Hello, I have installed WEC-Sim in  MATLAB versions (2017b, 2024b, 2020b, 2018b).
I have created a startup.m file and saved.

I set the paths to C:\Users\Enoch\Desktop\$WECSIM\WEC-Sim-6.1.2\source.

WEC-Sim appears in the simulink library. I have followed the installation steps.

I get error messages whenever I want to test the installation by typing "wecSim"

What could be the problem or cause of this?

Thank you

dforbush2, 2025-03-06:
Hi @EnochBadu 

Can you please share the error message you are getting? It appears you are using a static code download. After navigating to your directory

./WEC-Sim-6.1.2/

Please run 

`addWecSimSource`

And then navigate to an example, say ./Examples/RM3, and run

`wecSim`

If this runs, you have installed correctly.

EnochBadu, 2025-03-06:
Thank you for your response  *dforbush2, *
Below is the error message I get each time I run wecSim.

Initializing the Simulation Class...
Case Dir: C:\Users\Enoch\Documents\MATLAB
Error using simulationClass/checkInputs (line 129)
The simMechanics file, C:\Users\Enoch\Documents\MATLAB\WEC_1_NoGearbox.slx,
does not exist in the case
directory

Error in initializeWecSim (line 111)
simu.checkInputs();

Error in run (line 91)
evalin('caller', strcat(script, ';'));

Error in wecSim (line 40)
run('initializeWecSim');

I hope to hear from you soon.
Thank you

EnochBadu, 2025-03-06:
Added to this mail is a screenshot of my interface for your reference.

Thank you very much for your response and I hope to hear from you soon.

dforbush2, 2025-03-06:
Hello @EnochBadu 

Please review our documentation! http://wec-sim.github.io/WEC-Sim
It looks like you are not trying to run wecSim from a case directory. Your case directory should include your Simulink model and your WEC-Sim input file (if using). The file paths in your wecSimInputFile to your hydrodynamic data should be relative to your case directory. 

As a test, please navigate in MATLAB to ./Examples/RM3, and run `wecSim` from the command line. This should run, and is an example of the case directory.