WEC-Sim/WEC-Sim #1437: bug in "writeParaviewMooring"

hamzaJayy, 2025-03-12:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
Hey guys, I was running RM3 with moordyn. When I turned simu.paraview.option=1, and simu.paraview.dt =1 which is greater than simu.dt, I got the following error:

Index exceeds the number of array elements. Index must not exceed 41.

Error in writeParaviewMooring (line 60)
            pt = [moorDyn.(['Line' num2str(iline)]).(['Node' num2str(inode) 'px'])(it), moorDyn.(['Line' num2str(iline)]).(['Node' num2str(inode) 'py'])(it), moorDyn.(['Line' num2str(iline)]).(['Node' num2str(inode) 'pz'])(it)];

Error in paraviewVisualization (line 62)
        writeParaviewMooring(output.moorDyn, modelName, output.moorDyn.Lines.Time, datestr(simu.date), mooring.moorDynLines, mooring.moorDynNodes,simu.paraview.path,TimeBodyParav,NewTimeParaview)

Error in stopWecSim (line 38)
paraviewVisualization

Error in run (line 112)
evalin('caller', strcat(scriptStem, ';'));

Error in wecSim (line 56)
run('stopWecSim');

Upon looking inisde the "writeParaviewMooring" function, i found that the for loop that writes paraview files does not use "NewTimeParaview" vector instead it still uses "TimeBodyParav" which is original time vector.

I fixed this problem on my end by replacing "for it = 1:length(TimeBodyParav)" with "for it = 1:length(NewTimeParaview)".

additionally, the function "writeParaviewMooring" also interpolates Line#.Seg#Te which i dont think is required for paraview. It could be avoid to improve time. 

On second topic, is "writeParaviewMooring" for 6.1 slower than that of wecsim 6.0?

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: windows 11
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2024a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1

**WEC-Sim file(s):**
_none

**How to reproduce:**
run moordyn application with simu.paraview.dt >simu.dt

**Expected behavior**
bug in writeParaviewMooring causing it not to execute.

**Screenshots:**

[image]

**Additional information:**
Similar probelm occured with wecSim 6.0.  there I have to replace this line
 "NewTimeParaview(:,1) = simu.paraview.startTime:simu.paraview.dt:simu.paraview.endTime; "
with this line:
" NewTimeParaview = simu.paraview.startTime:simu.paraview.dt:simu.paraview.endTime;"

jtgrasb, 2025-03-12:
@hamzaJayy Thank you for sharing the issues you're encountering. By applying the changes you suggest in #1438, this error should be fixed because the `TimeBodyParav` variable is always set equal to `NewTimeParav`. 

The interpolation is necessary for the locations and tensions to output their values with the paraview output timestep. While this is not necessary in your case with a fixed timestep, it may be necessary when using a variable timestep.

Lastly, there were some changes to the paraview functions by #1081, but these were included in the v6.0 release. There have not been any significant changes to the paraview functions since then so I don't expect there to be large differences in computation time for outputting the paraview data in v6.1 vs. v6.0. The whole RM3_Moordyn model will likely run slower in v6.1 due to changes to MoorDyn though.

hamzaJayy, 2025-03-13:
@jtgrasb makes sense. Thank you.

closed 2025-03-13 by jtgrasb