WEC-Sim/WEC-Sim #1438: Bug in paraviewVisualization.m

hamzaJayy, 2025-03-12:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
The properties for simu.paraview to set the start, timestep and endtime are being defined inside the for loop of body. Thats means the "if isempty(simu.paraview.startTime) || isempty(simu.paraview.dt) || isempty(simu.paraview.endTime)"  logic only works for first body, never for other bodies. Then "writeParaviewBody" is being called with different "TimeBodyParav". 

[image]

I thik it should be like this

[image]

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html):  2024a
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html):  v6.1

**WEC-Sim file(s):**
paraviewVisualization.m

**How to reproduce:**
set simu.paraview.dt > simu.dt and run rm3 model with moordyn. Then in paraview, load the paraviewfiles, check the information of each file. First body will have different time vector than the rest. 

**Expected behavior**
in paraviewfiles, every entity has the same time vector

**Screenshots:**

[image]

[image]

**Additional information:**
same issue in wecsim 6.0

jtgrasb, 2025-03-12:
@hamzaJayy Thanks for catching this issue. I agree with your proposed solution. Are you able to create a pull request to the main branch to fix this bug? Please let me know and I could review it. If not, I can apply the fix myself.

hamzaJayy, 2025-03-13:
@jtgrasb I haven't done this before. I tried through dev branch, but it says I dont have authority to do so. Please go ahead and apply the fix. Thanks

jtgrasb, 2025-03-13:
No problem, I created PR #1441.

jtgrasb, 2025-04-16:
Resolved by PR #1441

closed 2025-04-16 by jtgrasb