WEC-Sim/WEC-Sim #1440: Usage of wecSim and MoorDyn v2 to model multiple boies

hamzaJayy, 2025-03-13:
**Describe the question (REQUIRED):*
I am trying model a simple case i.e, two floating bodies have a mooring line hanging at the bottom from each body. I am modelling both these bodies as coupled for moordyn. Schematic of model look like this: 

[image]

i am getting this error: 

[image]

I have tried so many things but not working. the line parameters are fine because the model ran fine when I modelled single mooring sytem. 
**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: windows 11
 - [MATLAB/Simulink Release] 2024a
 - [WEC-Sim Release] v6.1.1

**WEC-Sim file(s):**

[model_files.zip](https://github.com/user-attachments/files/19223253/model_files.zip)

**How to reproduce:**
I attaached the files to run the model.

**Expected behavior**
_Describe the expected behavior._

**Screenshots:**
WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory... 
WEC-Sim: An open-source code for simulating wave energy converters
Version: 6.1

Initializing the Simulation Class...
	Case Dir: D:\modelling\CETO\moorDyn2_test 
---------------Starting MoorDyn-----------
Warning: The function 'AllOutput' was not found in the library 
> In loadlibrary
In mooringClass/callMoorDynLib (line 185)
In initializeWecSim (line 150)
In run (line 112)
In wecSim (line 40) 
MoorDyn Initialized. Now time stepping...
Elapsed time is 4.086293 seconds.

WEC-Sim Pre-processing ...   
Elapsed time is 0.079308 seconds.

WEC-Sim Simulation Settings:
	Start Time                     (sec) = 0
	End Time                       (sec) = 10
	Time Step Size                 (sec) = 0.1
	Ramp Function Time             (sec) = 0
	 Number of Time Steps     = 100 

Wave Environment: 
	Wave Type                            = No Wave (Constant Hydrodynamic Coefficients)
	Hydro Data Wave Period, period (sec)    	= 4

List of Body: Number of Bodies = 0 

List of PTO(s): No PTO in the system

List of Constraint(s): Number of Constraints = 3 

	***** Constraint Name: Constraint1 *****

	***** Constraint Name: Constraint2 *****

	***** Constraint Name: Constraint3 *****

Simulating the WEC device defined in the SimMechanics model D:\modelling\CETO\moorDyn2_test\simpleModel.slx...   
Elapsed time is 0.157939 seconds.
Error using wecSim (line 43)
Size mismatch for MATLAB expression '<output of MoorDyn_caller>'. Expected = 1x12 Actual = 1x6

**Additional information:**
Once got it running, i would like to model two floating bodies and mooring line between them using moordyn. Something like this:

[image]

jtgrasb, 2025-03-14:
@hamzaJayy Thank you for creating the issue and sharing the model. I believe the issue is that the force returned by MoorDyn is being initialized as a 1x6 vector instead of 1x6*nBod. I have created and merged [PR 12](https://github.com/WEC-Sim/MoorDyn/pull/12) to the MoorDyn repository to resolve this issue. I expect this will also resolve the crashing of MATLAB. Please let us know if you still encounter issues.

hamzaJayy, 2025-03-14:
I read your comment in PR12. I think using moordyn connections to determine force vector might got into issue. There could scenario where you only need one mooring connection but two coupled bodies. Please refere to my 2nd sketch above. Am I right?

jtgrasb, 2025-03-14:
@hamzaJayy In any situation where you have one MoorDyn connection, MoorDyn will return a 1x6 force vector. The displacement vector that we input from WEC-Sim to MoorDyn should always be the same size as the force vector returned. Simulink will then handle applying that force vector (equal and opposite) to each connection point (i.e., both bodies).

hamzaJayy, 2025-03-14:
@jtgrasb yes force will be applied to both bodies but there will be only one coupled body. At the moment, wecsim can model two bodies as coupled (in input file for moordyn) and only one moordyn connection. 
I got it by having dummy free mass then connecting it with B of two moordyn connections. F of each moordyn connection is then connected with one coupled body. 

[image]

it is working. But issue is, "writeParaviewMooring" is not working. It only works if there is only one mooring connection. There is need of some work as well.

A-Musaad, 2025-03-16:
Hi @jtgrasb,
Thank you for updating the WecSim v6.1.2 source code to work with the multiple bodies with MoorDyn problem. 

**Another Issue:** I noticed that the simulation starts without MoorDyn converging on the initial mooring lines shape first as in previous versions. 
an initial mooring shape example below
[image]

RyanDavies19, 2025-03-17:
@A-Musaad, can you share the `options list` section of your MoorDyn input file? 

If you want to replicate the initialization of old MoorDyn versions, you need to add the `1           ICgenDynamic` flag to the options list. This will tell MoorDyn to use the old dynamic relaxation method to initialize. If you are still not seeing the results you want, you need to play with the initial condition steady state sensitivity.

jtgrasb, 2025-03-17:
@hamzaJayy I've also addressed the MoorDyn to Paraview workflow for multiple connections in PR #1441. Will update when reviewed and merged.

hamzaJayy, 2025-03-18:
Thank you @jtgrasb

A-Musaad, 2025-03-18:
Thanks @RyanDavies19,option  1           ICgenDynamic resolved my inquiry.

jtgrasb, 2025-04-16:
This issue will be resolved by PR #1455 when merged.

hamzaJayy, 2025-04-28:
@jtgrasb hey, I am back on this again. One potential issue is: the wecsim does not know which lines in moordyn are part of a particular mooring object. 

Forexample: I have two lines in lines.text as this:

[image]

And each mooring object is initialized like this:

[image]

how does wecsim ensures that line#1 is for mooring(1) and line#2 is for mooring(2)? 

Because line numbers inside responseClass.loadMoorDyn, iline are number from 1:linesNum for each mooring object.

[image]

As moorDyn ouputs  lines_Line##.out sequentially for all line, responseClass.loadMoorDyn should read/write data in a way that each line is assigned to its appropriate mooring object. 

Thank you

jtgrasb, 2025-04-28:
@hamzaJayy Thank you for bringing this to our attention. We assume that the lines are defined in the same order in the MoorDyn input file as the WEC-Sim input file. That is, all of mooring(1) lines are defined before all of mooring(2) lines. 

You are also correct with regards to the outputs. It is currently only set up to handle one instance of MoorDyn. I will update this within the existing PR. Thanks!

kmruehl, 2025-05-15:
Thanks @jtgrasb. @hamzaJayy  please reopen the issue if you have any other questions regarding this issue

closed 2025-05-16 by jtgrasb