WEC-Sim/WEC-Sim #1443: [Feature]Calculation reference point of QTF matrix

TianyuanWangi, 2025-03-14:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the requested feature (REQUIRED):*
_Provide a clear and concise description of the requested feature._

Hi, developers

I find that the calculation reference point of QTF matrix is the COG of the floating body. Will you add the additional option to specify a separate reference point for the 2nd order wave exciation calculation?

For the floating offshore wind turbines, the calculation reference point of QTF matrix is generally set as the COG of the whole system, but not that of the floating platform. Therefore, is it possible to add a coordinate transformation function to the calculation of second-order wave loads to deal with inconsistent calculation reference points. This may involve a coordinate transformation of motions and loads.

Best regards,

Tianyuan

**Related issue:**
_Does this feature request relate to an issue? If so, provide a clear and concise description of the issue. For example, "I'm try to [...]", and (if applicable) reference the related issue or pull request._

**Describe alternatives solutions:**
_Are there other possible solutions? If so, provide a clear and concise description of possible alternative solution(s)._

**Related  application:**
_Does this feature request relate to a specific application? If so, describe the application so the WEC-Sim team can better understand your goal(s)._

**Additional information:**
_Provide any other additional information._

TianyuanWangi, 2025-03-19:
Yes, my pleasure. But this can involve a lot of changes, which I'll briefly list here

- In bodyClass.m, Add a variable to specify the new body reference point (BRP) of 2-nd excitation load
- In WECSim_Lib_Body_Elements.slx, transform the displacement, velocity and acceleration at COG to BRP
- In WECSim_Lib_Body_Elements.slx, transform the 2-nd excitation load from BRP to COG
- The coordinate transformations need to be implemented using eulXYZ2RotMat or more accurate methods

Again, I am happy to contribute to WEC-Sim to the best of my ability.

Best regards,

Tianyuan

MShabara, 2025-03-25:
@TianyuanWangi 

1. I believe simulink will define the body reference frame location during the simulation, so no need to define it in the `bodyClass`.
2. The phase shift in force can be introduced similarly to the first-order excitation force, as shown in [this link](https://wec-sim.github.io/WEC-Sim/dev/user/advanced_features.html#large-x-y-displacements) for the `largeXYDisplacement` feature.
The trick is how to implement this phase shift into the [secondOrderExt.m](https://github.com/WEC-Sim/WEC-Sim/blob/main/source/functions/simulink/model/secondOrderExt.m) function.

3. And yes, coordinate transformations is needed.

The force correction is implemented in [OpenFAST](https://github.com/OpenFAST/openfast), but I have not checked the methodology they followed to implement it.

kmruehl, 2025-05-15:
Thanks @MShabara. @TianyuanWangi if you have any updates on this, please submit a pull request.

closed 2025-05-15 by kmruehl

MShabara, 2025-05-17:
Hi @TianyuanWangi, I’ve opened PR #1477 to address the handling of rotations.

There was a bit of a misunderstanding from my part in my previous comment—I was referring specifically to the large X-Y motions observed during simulations.

Regarding the calculation reference point for the QTF matrix: Yes, it is typically set at the center of gravity (COG) of the entire system, rather than just that of the floating platform. But isn't it how it's supposed to be?

TianyuanWangi, 2025-05-17:
Hi [MShabara](https://github.com/MShabara)

Actually, for the floating WECs, the calculation reference point for the QTF matrix is COG. But for the floating wind turbines, the reference point should be the total COG (TCOG) of the system, which is usually higher than than COG. 
Therefore, two solutions can be employed: the first is to modify the reference point of the frequency-domain hydrodynamic coefficients, and the second is to transform the reference point in time domain.

1. As far as I know, WAMIT supports separate specification of XBODY and XCG. For the wind turbines, XBODY should be the local coordinate origin (COG of the floating body) and XCG should be TCOG of the system. But AQWA doesn't have this feature.
2. The second method is to transform the reference point from COG of the floating body to TCOG. This method is relatively straightforward and feasible for the hydrodynamic coefficients obtained from different hydrodynamic software.

I hope this helps.