WEC-Sim/WEC-Sim #1444: [Support] Restricting WEC Motion to the Z-Direction in WEC-Sim

mech2809, 2025-03-17:
Hi,

I am trying to model a WEC with three paddles and a main hall. The hall is allowed to have free motion only in the Z-direction, moving along with the waves. The WEC is not moored, and the relative motion between the paddles and the hall will be used to extract power.

However, I am facing difficulties in properly modeling the WEC to ensure that the hall moves only in the Z-direction while keeping the paddles attached to it.

Could you provide guidance on the best approach to achieve strict vertical motion (heave-only) for the hall, while ensuring that the paddles remain attached and move relative to it?

Any suggestions or examples would be greatly appreciated!

Thanks in advance.

dforbush2, 2025-03-17:
Hi @mech2809 

The translations constraint or PTO will constrain motion to a single translational direction, which, by default is the Z-axis.

However, I am wondering if this makes sense with the WEC you describe. If your WEC is not moored, what is restricting the motion to just Z? I encourage you to look at the RM3 example case, which relies on the relative heave motion between 2 bodies to extract power. As you'll note the whole 2 body system is coupled to the sea-bed reference frame using a 3-DOF floating constraint, and the 2 bodies are coupled by the translational PTO. This means that as the 2-body system pitches, the _local_ Z-axis along which the PTO acts may not be aligned with the _global_ Z-axis, along which Gravity acts. This is shown in the second image: the long axis of the cylindrical spar has pitched relative to the global coordinate system (shown)

[image]

[image]

If you do want a 2-body WEC that ONLY moves in the vertical direction, I recommend using 2 translational constraints: one coupling the 2 body system to the seabed, and another coupling the bodies together. In this case, the _global_ Z will be constrained to always align with the _local_ Z.

mech2809, 2025-03-18:
Hi 
Thanks for your response.
What i am trying to simulate is a Autonomous Under water vehicle which can charge itself by using wave energy. 
Please check the image for the WEC. Now i am trying to model in such a way that the WEC should move along the waves in Z direction and the relative motion between the main central hall and the paddles produce power to charge the WEC. 
Also the main hall is not moored and not fixed. 
[image]

dforbush2, 2025-03-18:
the following model structure will serve your needs and will only move in the +/-Z direction. The paddles act through rotational constraints attached to the central (blue) body. The blue body is constrained via a translational constraint to move only in Z. 

[image]

However, I do invite you to consider _how_ the real, physical device will be constrained to only move in the Z direction without moorings or controls. It may be prudent to model the device under a more realistic motion constraint.

dforbush2, 2025-03-21:
Hi @mech2809 
did this answer your question? I will close this issue for now, but feel free to re-open if you continue to have questions about this issue.
Best,
Dom

closed 2025-03-21 by dforbush2

mech2809, 2025-03-26:
Hey Dom,
Thanks for such a helpful answer.
I wanted wanted to ask can we allow the WEC to move in XYZ direction, which constraint used be used to do so. 
Also only 3 DOF will be enough to model it more precisely or i should consider all 6 DOF.
Thanks

dforbush2, 2025-03-26:
Hi @mech2809 

I would recommend 6DOF for your device. Even with unidirectional waves, the reaction force against the motion of your paddles (i.e., equal and opposite their PTO force) will cause a torque that has components inducing accelerations out of the XY plane. To do this, replace the translational constraint(1) in the above screenshot with a Floating 6DOF constraint. 

Bear in mind that if you don't want your device to drift in surge or sway, I would introduce a mooring force (above the Floating constraint, but below body(4), to keep the device from drifting substantially. Specifying this mooring will also be a key design decision!

Good luck!