WEC-Sim/WEC-Sim #1456: [Support] Transformation of hydrodynamic data reference point between HydroD and WEC-Sim (BEMIO)

Will891121, 2025-04-06:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the question (REQUIRED):**  
I'm working on a multi-body simulation for the NREL Task 37 configuration using WEC-Sim, with three floating buoys. I'm using HydroD to calculate hydrodynamic coefficients and exporting WAMIT's output files for use with WEC-Sim (BEMIO).

However, I’ve encountered an important issue:

HydroD computes the hydrodynamic coefficients with respect to the **waterplane (z = 0)**, and this reference point cannot be changed. But WEC-Sim requires the hydrodynamic data to be calculated with respect to the **center of gravity (CoG)** of each body.

I understand that modifying the `cg` field in the `hydroData` structure is not the solution, since the hydrodynamic matrices like added mass, radiation damping, and excitation moments are still computed around the waterplane origin and would thus be incorrect if not transformed.

My questions are:

1. What is the **correct way to transform the hydrodynamic data** from the waterplane (z=0) reference point to the CoG for use in WEC-Sim?
2. Does BEMIO provide any built-in method or utility function to assist with this transformation?
3. If not, are there any guidelines or recommended practices for manually transforming the hydrodynamic matrices to match the CoG reference point?
4. I would like to confirm whether it is valid for each body in a multi-body system to use the same hydrodynamic `.h5` file in the `wecSimInputFile`. I noticed that in the `WEC-Sim_Applications` repository, the RM3 example includes two bodies that both reference the same hydrodynamic file.  

---

**Workstation information (REQUIRED):**  
- Operating System: Windows 11  
- MATLAB/Simulink Release: R2024b  
- WEC-Sim Release: v6.1.2  

---

**WEC-Sim file(s):**  

---

**How to reproduce:**  
Use HydroD to compute hydrodynamic coefficients with reference to the waterplane, then attempt to use these in WEC-Sim for a multi-body system where the CoG is far below the z=0 level.

---

**Expected behavior:**  
The simulation should properly account for the forces and moments acting about each body’s CoG.

---

**Screenshots:**  
N/A

---

**Additional information:**  

I sincerely appreciate your time and guidance. Looking forward to your advice.

jtgrasb, 2025-05-05:
@Will891121 Do you have further questions regarding this issue? If not, I will close the issue thread.

Will891121, 2025-05-08:
@jtgrasb  Thank you for your advice on this issue.

I’ve resolved the problem using BEMRosetta, which allowed me to transform the hydrodynamic coefficients from the water surface reference (z = 0) to the body-centered reference required by WEC-Sim.
I really appreciate your helpful support!

Best regards,
Will

jtgrasb, 2025-05-08:
@Will891121 Great to hear! I wasn't sure whether BEMRosetta had this feature. For our future reference, which functions did you end up using for the transformation?

Will891121, 2025-05-15:
[image]
Hi @jtgrasb 
The process is actually quite simple.
What I did was:
I loaded the hydrodynamic data file into BEMRosetta, and choose the Advanced tab, updating the Body Axis (shown in figure) for each body to match the desired reference point. And then I choose what kind of hydrodynamic files I prefered to output.

I've attached a screenshot showing how this can be done.

Hope this helps others facing the same issue!

 Thanks for your help again!

Best regards,
Will

closed 2025-05-15 by jtgrasb