WEC-Sim/WEC-Sim #1462: [Support] PTO damping value in OSWEC example

fanofdw, 2025-04-18:
**Please review the following prior to submitting an issue:**
1. Read through the [WEC-Sim Documentation](https://wec-sim.github.io/WEC-Sim), especially information in the [Advanced Features](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html) section.
2. Review WEC-Sim ([source code](https://github.com/WEC-Sim/WEC-Sim/tree/main/source)
3. Review [WEC-Sim examples](https://github.com/WEC-Sim/WEC-Sim/tree/main/examples) and [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications).
4. Review prior [WEC-Sim issues](https://github.com/WEC-Sim/WEC-Sim/issues), search past issues and discussions that may help answer your question.

**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the question (REQUIRED):*

Hello, WEC-Sim Team!

WEC-Sim is an excellent tool for WEC simulation. I would like to express my gratitude for your dedicated work.

I encountered a problem when setting the PTO damping value. Taking the OSWEC code as an example, I utilized a regular wave (H = 2.5 m, T = 8 s) to facilitate a better comparison and set the PTO damping at c = 2.0E+07 Ns/m/rad.

I observed that the amplitude of body (1).forceTotal is even smaller than that of pto.forceTotal (as Figure 1). The output power is larger than the incoming wave power (55794.6 W/m × 18 m = 1004.3 kW) (as Figure 2).

From an energy conversion perspective, the energy of wave excitation is converted into the mechanical energy of the flap, the radiation wave energy, the energy absorbed by the PTO system, and the energy dissipated by viscous damping. Therefore, shouldn't the pto.forceTotal be smaller than the body (1).forceTotal? I speculate that this discrepancy may be due to the PTO damping being set to an unreasonably high value. And, for some reason, the WEC-Sim does not terminate the computation and instead produces a misleading result.

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 10
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2021b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1

**WEC-Sim file(s):**
https://github.com/WEC-Sim/WEC-Sim/tree/main/examples/OSWEC

**How to reproduce:**
Use regular wave (H = 2.5 m, T = 8 s) and set the PTO damping at c = 2.0E+07 Ns/m/rad.

**Expected behavior**
The pto.forceTotal should be smaller than the body (1).forceTotal.

**Screenshots:**

[image]

[image]

**Additional information:**
/

jniffene, 2025-05-14:
Hi thank you for reaching out! To fully assess the energy conservation here I examined the rotational work from the PTO and the body of the flap. That work is equivalent to the torque times the angular displacement $W = \tau\theta$. For the flap there are three torques acting on it, the rotational component acting over the center of gravity of the device and the torques due to the x and z force components, see below for a diagram of the body torques.

[image]

Below is a comparison of the flap and PTO rotational work. Note that the flap work is slightly higher than the PTO work, likely due to the high damping value.

[image]

I hope this provides some clarity on this!

fanofdw, 2025-05-16:
@jniffene Thank you for your prompt and detailed response. It seems I had previously confused the term body.forceTotal with 
τbody. I now realize that body.forceTotal actually represents τry (first part of τbody), and it is specifically centered around the flap's center of gravity, rather than the hinge point. Based on this clarification, the PTO damping coefficient can indeed be set to  c = 2.0E+07 Nsm/rad. Have I understood this correctly?

jniffene, 2025-05-16:
Yes that is correct!