WEC-Sim/WEC-Sim #1463: [Support] Multiple Gotos when running

tranlong932004, 2025-05-03:
I am a new user Matlab and have the following problem:

[image]
[image]
[image]

Any idea why this could be occurring? Thanks in advance!

jleonqu, 2025-05-13:
Hello @tranlong932004 
Could you please re open this issue using the GitHub template to provide more details so we can better help you?

jleonqu, 2025-05-13:
I was checking this really quick, and the problem is that you are using twice the block "Three-Phase V-I Measurement" without changing the labels in the blocks. You need to use different labels if you are using this block many times. This is where you can change the name:

[image]

kmruehl, 2025-05-15:
Thanks @jleonqu. I'm going to go ahead and close this issue. @tranlong932004 please reopen if you have any additional questions.

closed 2025-05-15 by kmruehl