WEC-Sim/WEC-Sim #1471: [Developer Issue] Fix regression tests for RM3

jleonqu, 2025-05-15:
There are some issues related to the regression tests for the RM3 example.

1. The first issue is that the files that are being used as reference to compared and verify the wecSim outputs don't have the required data for the comparison.
2. The tests were run to compare results from different WEC-Sim versions against the current development branch. Many tests are currently failing. The versions compared were v4.4 and v6.1.2. The timeline of the versions that were compared and their possible cause for failing are shown in the next timeline:

[image]

We need to:

1. Check why the tests are failing. The issue may be related to very tight tolerances in the tests.
2. Update the reference files to ensure they include the required data for accurate comparison with current WEC-Sim outputs.

cornerzdc, 2025-06-01:
why my  regressionTest is error ？

wecSimTest
Running bemioTest
......
Completed bemioTest

Running regressionTest

WEC-Sim Input From Standard wecSimInputFile.m Of Case Directory...
WEC-Sim: An open-source code for simulating wave energy converters
Version: 6.1.2

Initializing the Simulation Class...
Case Dir: F:\academic\code\simiturbine_oc2025\WEC-Sim-main\WEC-Sim-main\tests\RegressionTests\RegularWaves\regular
Elapsed time: 0.272462 seconds.

WEC-Sim Pre-processing...
Infinite water depth specified in BEM and "waves.waterDepth" not specified in input file.
Set water depth to 200m for visualization.
Elapsed time: 0.220354 seconds.

WEC-Sim Simulation Settings:
Start Time (sec) = 0
End Time (sec) = 200
Time Step Size (sec) = 0.1
Ramp Function Time (sec) = 100
Number of Time Steps = 2000

Wave Environment:
Wave Type = Regular Waves (Constant Hydrodynamic Coefficients)
Wave Height, height (m) = 2.5
Wave Period, period (sec) = 8

List of Body:
Number of Hydro Bodies = 2

***** Body Number 1, Name: float *****  
Body CG                          (m) = [0,0,-0.72]  
Body Mass                       (kg) = 725833  
Body Moments of Inertia       (kgm2) = [2.09073E+07,2.13061E+07,3.70855E+07]  
Body Products of Inertia      (kgm2) = [0,0,0]  

***** Body Number 2, Name: spar *****  
Body CG                          (m) = [0,0,-21.29]  
Body Mass                       (kg) = 886687  
Body Moments of Inertia       (kgm2) = [9.44196E+07,9.44071E+07,2.85422E+07]  
Body Products of Inertia      (kgm2) = [0,0,0]  
List of PTO(s): Number of PTOs = 1

***** PTO Name: PTO1 *****  
PTO Stiffness           (N/m;Nm/rad) = 0  
PTO Damping           (Ns/m;Nsm/rad) = 1.2E+06  
List of Constraint(s): Number of Constraints = 1

***** Constraint Name: Constraint1 *****  
Simulating the WEC device defined in the SimMechanics model F:\academic\code\simiturbine_oc2025\WEC-Sim-main\WEC-Sim-main\tests\RegressionTests\RegularWaves\regular\RM3.slx...
Elapsed time: 0.257999 seconds.

================================================================================
Error setting up or tearing down regressionTest.
As a result, all regressionTest tests failed and did not complete.
--------
Error ID:
--------
'MATLAB:MException:MultipleErrors'
-----------
Error details:
-----------
Error using wecSim (line 43)
Error due to multiple causes.

Error in runLoadRegular (line 24)  
wecSim;                     % Run Simulation  

Error in regressionTest/runRegTest (line 49)  
            runLoadRegular;  

Cause:  
    Error using wecSim (line 43)  
    Due to errors in the block body or limitations of the underlying analysis, Simulink cannot determine the output size and/or type of the block 'RM3/Rigid Body1/Hydrodynamic Body/Wave Diffraction and Excitation Force Calculation/Linear Wave  
    Excitation Force Variant Subsystem/Regular Wave  Excitation Force/MATLAB Function1'. The error may be inaccurate. Please fix the indicated errors or explicitly specify the size  
    and/or type of all block outputs.  

    Error using wecSim (line 43)  
    Error in port widths or dimensions. 'Output Port 1' of 'RM3/Rigid Body1/Hydrodynamic Body/Wave Diffraction and Excitation Force Calculation/Linear Wave  
    Excitation Force Variant Subsystem/Regular Wave  Excitation Force/MATLAB Function1/disp' has 2 elements. This port does not accept the dimensions (or  
    orientation) specified by the output signal.  
================================================================================

[Additional similar error sections translated with the same pattern...]

================================================================================
Error setting up or tearing down regressionTest.
As a result, all regressionTest tests failed and did not complete.
--------
Error ID:
--------
'MATLAB:undefinedVarOrClass'
-----------
Error details:
-----------
Cannot resolve the name 'regular.B1.WEC_Sim_new.time'.

Error in printPlotRegular (line 33)  
m = plot(regular.B1.WEC_Sim_new.time,regular.B1.WEC_Sim_new.heave,'k:',...  

Error in regressionTest/plotRegTests (line 98)  
                printPlotRegular  
================================================================================

Completed regressionTest

Running runFromSimTest
[Additional sections translated similarly...]

Failure Summary:

 Name                                        Failed  Incomplete  Reason  
=======================================================================  
 regressionTest/body1_reg_disp_heave           X         X       Error.  
-----------------------------------------------------------------------  
 regressionTest/body2_reg_disp_heave           X         X       Error.  
-----------------------------------------------------------------------  
 [Additional failure entries translated similarly...]  
MATLAB code coverage report saved to:
C:\Users\111\AppData\Local\Temp\tp207c876a_39bf_43f3_b7ac_1aab6e1e290a\index.html

ans =

52×6 table

[Table content translated similarly...]

ans =

1×52 TestResult array - Properties:

Name  
Passed  
Failed  
Incomplete  
Duration  
Details  
Summary:
31 Passed, 21 Failed (rerun), 21 Incomplete.
219.368 seconds testing time.

kmruehl, 2025-06-11:
resolved with https://github.com/WEC-Sim/WEC-Sim/pull/1473

closed 2025-06-11 by kmruehl