WEC-Sim/WEC-Sim #1495: Questions about using WEC-Sim MOST&MoorDyn

wind2021, 2025-07-02:
**Please review the following prior to submitting an issue:**
1. Read through the [WEC-Sim Documentation](https://wec-sim.github.io/WEC-Sim), especially information in the [Advanced Features](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html) section.
2. Review WEC-Sim ([source code](https://github.com/WEC-Sim/WEC-Sim/tree/main/source)
3. Review [WEC-Sim examples](https://github.com/WEC-Sim/WEC-Sim/tree/main/examples) and [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications).
4. Review prior [WEC-Sim issues](https://github.com/WEC-Sim/WEC-Sim/issues), search past issues and discussions that may help answer your question.

**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the question (REQUIRED):*
_Provide a clear and concise description of the requested support. For example, "I'm try to [...]", and (if applicable) reference the related issue or pull request._

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): [e.g., 2022b, 2023a, 2023b, 2024a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): [e.g., v5.1, v6.0, v6.1]

**WEC-Sim file(s):**
_Identify the WEC-Sim file(s) with bug(s) that cause error(s), and upload relevant file(s) to help reproduce the error(s)._

**How to reproduce:**
_Describe the steps to reproduce the error:_

**Expected behavior**
_Describe the expected behavior._

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Additional information:**
_Provide any other additional information._

kmruehl, 2025-07-02:
@wind2021  can you please amend this issue template to include details about your question with WEC-Sim/MoorDyn?

kmruehl, 2025-07-02:
Thank you @wind2021!

wind2021, 2025-07-08:
Hi kmruehl! Can you please help me check the above-mentioned problems?

wind2021, 2025-07-10:
Dear  jtgrasb, 

>Thank you so mcuh for your suggestions and help! All of this was indeed caused by the abnormal initialization settings of Moordyn. After carefully reading the relevant documentation and adjusting the initialization settings, the aforementioned problem was resolved. 

> Kind regards,  Lei

closed 2025-07-10 by wind2021