WEC-Sim/WEC-Sim #1496: [BUG]

MShabara, 2025-07-03:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
The structure block within the hydro rigid body library has some parameters changed from main to dev branch

The dev branch has 

[image]

The main branch has

[image]

Was this change intentional?

akeeste, 2025-07-08:
@MShabara yes these changes are intentional and related to #1465

kmruehl, 2025-07-24:
@jtgrasb and @MShabara are working on debugging this issue.

closed 2025-07-28 by MShabara

MShabara, 2025-07-28:
@akeeste The model exhibiting the issue was built using an older version of MATLAB and requires some updates before we can properly test it.

I’ve done some independent checks, and WEC-Sim appears to be functioning as expected. Until we update the model and confirm whether the issue is due to the model itself and not WEC-Sim, I don’t see a reason to keep this issue open.

kmruehl, 2025-07-28:
@MShabara  what version of MATLAB?