WEC-Sim/WEC-Sim #1498: Help with Force Balance

jwills801, 2025-07-08:
**Is your question request related to a problem? Please describe.**
I do not understand why I can't get mass time acceleration to give the sum of forces. I get turned around easily in Simscape and the added mass is tricky there. I've tried many combinations of forces but can't find any that give structural mass times acceleration. 

**Describe the theory or implementation approach that you have a question**
I am using the OSWEC example code as a starting place. I am usually focused on the PTO, but have set that force to zero for the sake of debugging my understanding of the dynamics.

**Personal project context**
I want to design a MPC controller for a unique PTO that is not capable of continuous forces. To do this, I'd like to start with a state space representation of the OSWEC.

**Describe the type of conclusion or resolution you are looking for**
I'd like help figuring out what forces I need to include to accurately represent the ma=F that happens in simscape.

**Additional context**
This is the code I am running after running wecSim:
```
acceleration = output.bodies(1).acceleration';
forceTotal = output.bodies(1).forceTotal';
forceConstraint = output.constraints(1).forceConstraint';
forcePTO = output.ptos(1).forceTotal';
realforceTotal = +forceTotal-forcePTO-forceConstraint;

M = [eye(3,3)*body(1).mass, zeros(3,3); zeros(3,3), diag(body(1).inertia)];
massTimesAcceleration = M*acceleration;
i = 5; % look at just pitch for simplicity
figure, plot(time,massTimesAcceleration(i,:),time,realforceTotal(i,:)),legend('m*a','sum of Forces'), grid, xlabel('Time [s]'), ylabel('Torque [Nm]')
```
and this is the result:
[image]

Thank you! I really appreciate the support you all provide.

MShabara, 2025-07-08:
@jwills801 Thank you for the question,

Just a quick clarification — in dynamics, we don’t conserve forces or torques; instead, we conserve energy (or sometimes momentum, depending on the context). The answer to Issue #1462 might be helpful to you.

Let me know if you need further clarification

jwills801, 2025-07-22:
Hello @MShabara, thank you so much for your help. That was indeed the problem. I needed to include the contributions on the torque from the x and z directions: T_net = T_y + r*F_x*cos(theta) - r*F_z*sin(theta)

I had been doing that for all the forces, except it also needs to be done for the mass*acceleration term. When I do that, I get exact matching of mass times acceleration and the net torque.

Thank you very much for your help!