WEC-Sim/WEC-Sim #1501: [Support]  Unusually High Efficiency in PTO-Sim: Realistic Behavior or Input Issue?

harrypotter-17i, 2025-07-10:
**Please review the following prior to submitting an issue:**
1. Read through the [WEC-Sim Documentation](https://wec-sim.github.io/WEC-Sim), especially information in the [Advanced Features](http://wec-sim.github.io/WEC-Sim/man/advanced_features.html) section.
2. Review WEC-Sim ([source code](https://github.com/WEC-Sim/WEC-Sim/tree/main/source)
3. Review [WEC-Sim examples](https://github.com/WEC-Sim/WEC-Sim/tree/main/examples) and [WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications).
4. Review prior [WEC-Sim issues](https://github.com/WEC-Sim/WEC-Sim/issues), search past issues and discussions that may help answer your question.

**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the question (REQUIRED):*

Hello,

I'm new to WEC-Sim and have recently started exploring PTO-Sim for simulating hydraulic power take-off systems. During my initial simulations, I encountered some unexpected results regarding wave-to-mechanical energy conversion efficiency, and I’m trying to understand whether this is due to a mistake on my part or an issue with the model parameters.

I used the RM3_Hydraulic example provided in the WEC-Sim repository. I ran a simulation for 400 seconds and calculated the mechanical power output from the hydraulic motor and compared it to the incident wave energy over that period.

Here is the code I used to compute the efficiency:

***********************************************************************************************************************************
% Hydraulic motor outputs
shaftSpeed_rpm = output.ptoSim(5).shaftSpeed;
shaftSpeed_rad = shaftSpeed_rpm * (2*pi / 60);  % convert RPM to rad/s

motor_torque = output.ptoSim(5).torque;
mechanical_power = motor_torque .* shaftSpeed_rad;

motor_speed = output.ptoSim(5).shaftSpeed;      % Motor shaft speed (rad/s)

% Extract current and voltage from the electric generator (ptoSim(6))
current = output.ptoSim(6).current;    % Generator current (A)
voltage = output.ptoSim(6).voltage;    % Generator voltage (V)
time    = output.ptoSim(6).time;       % Time vector (s)

% Calculate electrical power (W)
electrical_power = abs(current .* voltage);

T1 = 0;
T2 = 400;

Wave_Energy = waves.power*20*400;  % Wave Power * length * time

idx = 1:length(output.ptoSim(1).time);
Total_MechEnergy = trapz(output.ptoSim(1).time(idx), abs(mechanical_power(idx)));
fprintf('Total Wave Energy = %0.3f\n', Wave_Energy);
fprintf('Total Mechanical Energy = %0.3f\n', Total_MechEnergy);
fprintf('Efficiency (Wave to Mech) = %0.3f %\n', Total_MechEnergy/Wave_Energy);
***********************************************************************************************************************************

And this was my output

***********************************************************************************************************************************
Total Wave Energy = 164105044.191
Total Mechanical Energy = 8283170.512
Efficiency (Wave to Mech) = 0.050 
***********************************************************************************************************************************

In the next case, I modified the wave input to match the sea state near my coastal area. Just the wave input - nothing else was changed. 

***********************************************************************************************************************************
%% Wave Information
%Irregular Waves using PM Spectrum
waves = waveClass('irregular');
waves.height = 0.86;
waves.period = 5.145;
waves.spectrumType = 'PM';
waves.phaseSeed=1;
***********************************************************************************************************************************

And this was my output

***********************************************************************************************************************************
Total Wave Energy = 12479945.315
Total Mechanical Energy = 5149934.450
Efficiency (Wave to Mech) = 0.413 
***********************************************************************************************************************************

1. Why is there such a drastic increase in efficiency just by reducing wave height and period?
 - Is this kind of sensitivity expected?
 - Does the system somehow perform more efficiently at lower wave energy input?

2. I noticed that shaft speed values in ptoSim(5).shaftSpeed seems to be in RPM. Using them directly in power calculations (without converting to rad/s) gives efficiency values greater than 1, which obviously isn't correct. Can you confirm that RPM is the correct interpretation?

3. Some parameter values in the RM3_Hydraulic input file seem unrealistically large, such as a 70 m stroke length. Is this value meant to represent a theoretical or idealized case? There were some issues with some other quantities also.
 - Should these default values be considered physically realistic, or are they primarily for demonstrating functionality?

I don't know if this is normal or if I made a mistake. Please shed some light on it

Workstation information (REQUIRED):
_Describe the workstation:_
 - Operating System: [Windows 11]
 - MATLAB/Simulink Release (https://www.mathworks.com/help/matlab/release-notes.html):  [ 2025a]
 - WEC-Sim Release (http://wec-sim.github.io/WEC-Sim/man/release_notes.html):  [Version: 6.1.2]

**WEC-Sim file(s):**
_Identify the WEC-Sim file(s) with bug(s) that cause error(s), and upload relevant file(s) to help reproduce the error(s)._

[Files.zip](https://github.com/user-attachments/files/21154672/Files.zip)

**How to reproduce:**
Run both the files, first the original wecsiminput file and then the modified file. Use the userdata provided for the post processing

**Expected behavior**
You will see that there is a drastic difference between the efficiency in the two cases

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screen shot of the exact error message from the MATLAB command line._

**Additional information:**
_Provide any other additional information._

harrypotter-17i, 2025-07-15:
It will be really helpful if you could give some updates on this

jleonqu, 2025-07-15:
Hello @harrypotter-17i 

1. The energy output is influenced by many factors, including the wave input, the PTO design, and the buoy design (mass, geometry, hydrodynamic characteristics, etc.). You are calculating the wave-to-wire efficiency, which generally tends to be low. The efficiency can improve significantly if the PTO is highly efficient at converting the mechanical input from the buoy into electrical power. However, for hydraulic systems, this is often not the case, as poorly designed hydraulic systems can have very low efficiencies. I recommend reviewing some wave-to-wire models to better understand how the different subsystems affect the overall performance of the device.

2. The output shaft speed is given in RPM, but for power calculations, you need to use rad/s.

3. The values in the example are not realistic; they are provided only to demonstrate the functionality. You should use realistic values that represent your specific case.

I’ll leave some recommended papers for you to review:

* An Efficient and Effective WEC Power Take-Off System: https://ieeexplore.ieee.org/abstract/document/10017387 
* Design and Performance Evaluation of a Resistive Control Using a Hydraulic PTO System for the TALOS Wave Energy Converter: https://onepetro.org/ISOPEIOPEC/proceedings-abstract/ISOPE24/All-ISOPE24/546405 
* Development of PTO-Sim: A Power Performance Module for the Open-Source Wave Energy Converter Code WEC-Sim: http://energy.sandia.gov/wp-content/uploads/2014/06/SAND2015-2069C.pdf
* PTO-Sim: Development of a Power Take Off Modeling Tool for Ocean Wave Energy Conversion: https://energy.sandia.gov/wp-content/uploads/2014/06/2015-IEEE-PES_PTO-Sim_Nak.pdf

I hope this helps!