WEC-Sim/WEC-Sim #1505: [BUG] Flex Body

kmruehl, 2025-07-15:
**Describe the bug (REQUIRED)**
The OWC/OrificeModel application no longer running. @MShabara resolved this issues in [#81
](https://github.com/WEC-Sim/WEC-Sim_Applications/pull/81), but the case is now failing. We think this could be related to either 

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: [e.g., iOS8.1, Windows 10, Linux 5.12.1]
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): [e.g., 2022b, 2023a, 2023b, 2024a]
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): [e.g., v5.1, v6.0, v6.1]

**WEC-Sim file(s):**
_What file(s) cause the error? Identify the WEC-Sim file(s) that cause the error, and (if applicable) upload relevant file(s)._

**How to reproduce:**
_How can we reproduce the error? Describe the steps to reproduce the error._

**Expected behavior**
_What is the excepeted behavior? Describe the expected behavior._

**Screenshots:**
_Provide relevant screenshots to help explain the problem. For example, a screenshot of the error message from the MATLAB command window._

**Additional information:**
_Provide any other additional information._

jtgrasb, 2025-07-15:
Updating after further investigation. I believe there are 2 bugs present:
1. The outport of the convolution integral calculation for the flex body is called "1", but it should be called "Out1" to be consistent with the variant subsystem terminology. This should resolve errors running models with the flex body such as the Generalized_Body_Modes application. 
[image]

2. The 2nd bug is specific to the OWC/OrificeModel. Inside the Flex Body, the "Wave Radiation Forces Calculation" subsystem is linked to the Rigid Body block, but this link is broken because the flex body handles the wave radiation forces calculation slightly differently. When looking at the OWC/OrificeModel model file (OWC_GBM.slx), because the 'Wave Radiation Forces Calculation' link to Rigid Body is broken, it is not being linked at all and has fallen behind. A quick fix for this is just to update the OWC_GBM.slx manually to make sure it is all caught up with the Flex Body and maintain the Orifice Model portion. However, I think its also worth considering an option that links the subsystems themselves to avoid these errors in the future. For example, instead of the Flex Body's "Wave Radiation Forces Calculation" subsystem being a broken link to Rigid Body, it could be fully unlinked allowing the OWC_GBM.slx's "Wave Radiation Forces Calculation" subsystem to be linked accordingly. 

[image]

@akeeste Since you are working on #1474 for library updates, do you want to incorporate bug fix 1 in your PR. And does #1474 also include a fix for bug 2? Or would that be something to consider in a later PR?

akeeste, 2025-07-16:
@jtgrasb yes I can add bugfix #1 to #1474. #2 will be addressed in a follow-on PR once I merge #1474, hopefully today

kmruehl, 2025-07-23:
@jtgrasb  please resolve #2 with a direct commit to applications dev

jtgrasb, 2025-07-23:
Issue 2 has been resolved by [Applications PR 92](https://github.com/WEC-Sim/WEC-Sim_Applications/pull/92).

closed 2025-07-23 by jtgrasb