WEC-Sim/WEC-Sim #1512: [BUG] Error on dev when simu.cicDt is different than simu.dt

jtgrasb, 2025-07-24:
**Note: Please complete all required fields. Italicized text describes each field.**

**Describe the bug (REQUIRED)**
When setting simu.cicDt as a value different than simu.dt, the interpolation of the convolution integral seems to cause issues. Here's the error I'm getting:
```
Error using wecSim
The data abscissae must be distinct and strictly monotonic.
	Error in interp1.m (line 208)
		            coder.internal.error('Coder:toolbox:interp1_nonMonotonicX');
	Error in interp1.m (line 54)
	Error in extrapFrad.m (line 29)
	Error in 'RM3/Float/Hydrodynamic Body/Wave Radiation Forces Calculation/SS CI and Constant-Damping-CoeVariant Subsystem/Convolution Integral Calculation/Convolution Variant
    Subsystem/Convolution Integral Calculation/MATLAB Function' (line 3)
```

**Workstation information (REQUIRED):**
_Describe the workstation:_
 - Operating System: Windows 11
 - MATLAB R2023b
 - WEC-Sim dev branch 

**WEC-Sim file(s):**
Seems to be an issue with how `convolutionIntegralInterp` and `extrapFrad` handle the timesteps (specifically when simu.cicDt is different than simu.dt).

**How to reproduce:**
Modify the RM3 case by setting `simu.cicDt = 0.2` and enable regularCIC waves.

**Expected behavior**
Should run without error.

**Screenshots:**
N/A 

**Additional information:**
N/A

akeeste, 2025-07-29:
Ah okay so this was actually a rate transition issue. The `timeExtrap` input to `extrapFrad` needs to be initialized a certain way. I believe @TianyuanWangi and I had this implemented before, but maybe it got lost in other library changes. Either way, I think I have a solution and will make a PR shortly

akeeste, 2025-07-30:
Resolved by #1518