WEC-Sim/WEC-Sim #1524: [Support] Help Running Example Capytaine Files

jwills801, 2025-08-07:
**Describe the question (REQUIRED):*
I am trying to rerun the capytaine example files for the OSWEC and compare the resulting .nc file to the one that already exists in the repository.

**Workstation information (REQUIRED):**
 - Operating System: Ubuntu 22.04.5 LTS
 - [MATLAB/Simulink Release](https://www.mathworks.com/help/matlab/release-notes.html): 2024b
 - [WEC-Sim Release](http://wec-sim.github.io/WEC-Sim/man/release_notes.html): v6.1

**WEC-Sim file(s):**
WEC-Sim/examples/BEMIO/Capytaine/oswec/oswec.py
WEC-Sim/examples/BEMIO/CAPYTAINE/call_capytaine.py

**How to reproduce:**
Run WEC-Sim/examples/BEMIO/Capytaine/oswec/oswec.py
(side note, I had to change the line which saves the .nc file - is this an equivalent expression? I was getting a type error with the original line of code.)
```
# cpt.io.xarray.separate_complex_values(capyData).to_netcdf(ncFName, encoding={'radiating_dof': {'dtype': 'U'}, 'influenced_dof': {'dtype': 'U'}})
    data = cpt.io.xarray.separate_complex_values(capyData)
    data["radiating_dof"] = data["radiating_dof"].astype(str)
    data["influenced_dof"] = data["influenced_dof"].astype(str)
    data.to_netcdf(ncFName)
```
After capytaine has been run, read it in matlab and compare it with the preexisting oswec_full.nc
```
hydro_old = struct();
hydro_old = readCAPYTAINE(hydro_old,'oswec_full.nc');
[w,w_ind] = min(abs(hydro_old(1).w-10));
disp('Exisiting Damping Coefficients at 10 rad/s')
hydro_old(1).B(1:6,1:6,w_ind)

disp('___________________________________________________')

hydro_new = struct();
hydro_new = readCAPYTAINE(hydro_new,'oswec.nc');
[w,w_ind] = min(abs(hydro_new(1).w-10));
disp('New Damping Coefficients at 10 rad/s')
hydro_new(1).B(1:6,1:6,w_ind)
```

**Expected behavior**
I expected these two matrices to be the same. Is that expectation correct?

**Screenshots:**
[image]

**Additional information:**
N/A

MShabara, 2025-08-10:
@mancellin
I ran the same model and obtained the following results:
[image] [image]

The newer version seems cleaner, the yellow plot was generated about two years ago, but I don’t recall which Capytaine version was used at that time.

jwills801, 2025-08-12:
Thank you both for your help and support!

closed 2025-08-12 by jwills801