WEC-Sim/WEC-Sim #1543: Applying Drag Coefficient using Morison Element

LavaZohreh, 2025-08-28:
Hi! 

[image]

I have a single body, which includes 4 vertical columns, with 6 m distance between them.  I'd like to add a Drag Coefficient using the Morison element. As I have seen in the form that Cd will be applied in the centre of gravity automatically, the distribution should be defined and I defined it for my case. I would be grateful if you could check this code and let me know if I defined it correctly. My below water body is shown in the attachment.

%% Add Drag
ZCOG = 0.7;      % Centre of Gravity
ZB1=-0.08375;    % Z of Morison Element_Level 1 --> the closest to water surface
ZB2=-0.25125;    % Z of Morison Element_ Level 2
ZB3=-0.41875;    % Z of Morison Element_Level 3
ZB4=-0.58625;    % Z of Morison Element _Level 4
ZB5=-0.75375;    % Z of Morison Element _Level 5
ZB6=-0.92125;    % Z of Morison Element _Level 6
ZB7=-1.08875;    % Z of Morison Element _Level 7
ZB8=-1.25625;    % Z of Morison Element _Level 8

area_side_X = 0.1675*5.224; % 
area_side_Y = 0.1675*1.835; % 

% 4 locations
rgME_array = [
    6   6  ZB1-ZCOG;        %First Column
    6   6  ZB2-ZCOG;
    6   6  ZB3-ZCOG;
    6   6  ZB4-ZCOG;
    6   6  ZB5-ZCOG;
    6   6  ZB6-ZCOG;
    6   6  ZB7-ZCOG;
    6   6  ZB8-ZCOG;
    6  -6  ZB1-ZCOG;        %Second Column    
    6  -6  ZB2-ZCOG;
    6  -6  ZB3-ZCOG;
    6  -6  ZB4-ZCOG;
    6  -6  ZB5-ZCOG;
    6  -6  ZB6-ZCOG;
    6  -6  ZB7-ZCOG;
    6  -6  ZB8-ZCOG;
   -6   6  ZB1-ZCOG;        %3rd Column
   -6   6  ZB2-ZCOG;
   -6   6  ZB3-ZCOG;
   -6   6  ZB4-ZCOG;
   -6   6  ZB5-ZCOG;
   -6   6  ZB6-ZCOG;
   -6   6  ZB7-ZCOG;
   -6   6  ZB8-ZCOG;
   -6  -6  ZB1-ZCOG;        %4th Column
   -6  -6  ZB2-ZCOG;
   -6  -6  ZB3-ZCOG;
   -6  -6  ZB4-ZCOG;
   -6  -6  ZB5-ZCOG;
   -6  -6  ZB6-ZCOG;
   -6  -6  ZB7-ZCOG;
   -6  -6  ZB8-ZCOG ];

% Repeat same values for each location
cd_array    = repmat([1 1 0], 32, 1);            % 32×3
ca_array    = zeros(32, 3);                      % 32×3 % added mass ratio: it has been already in included through imported data fron nemoh
area_array  = repmat([area_side_X area_side_Y 0], 32, 1);  % 4×3
VME_array   = zeros(32,1);                       % 32×1
z_array     = repmat([1 1 0], 32, 1);            % 32×3

body(1).morisonElement = struct( ...
    'option', 1, ...
    'cd', cd_array, ...
    'ca', ca_array, ...
    'area', area_array, ...
    'VME', VME_array, ...
    'rgME', rgME_array, ...
    'z', z_array);

DavideIssoglio, 2025-08-29:
Hello, at first sight, I would say that z_array is incorrect. I would do z_array = repmat([0 0 1], 32, 1); % 32×3, so that the axis of the slice is oriented along z. Also, since it seems you want to use the “normal+tangential” method and not “x-y-z” one (see body(ii). morisonElement.option  = 2 in https://wec-sim.github.io/WEC-Sim/main/user/advanced_features.html#morison-elements), pay attention to the correct definition. For example, if instead of your shape you had cylindrical slices, the normal area would be diameter*delta_z while the tangential area would be pi*diameter*delta_z. The rest seems correct to me (assuming that ZCOG is the centre of mass of the body).

LavaZohreh, 2025-09-01:
Thank you to @MShabara  and @DavideIssoglio for answering my question. Honestly, I used option 1 and wrote the code based on that. the results didn't seem correct to me. But I think it is better to switch to option 2 based on the geometry, which I think is more proper for a curved shape geometry.